intel-aero/meta-intel-aero #148: Replace initrd installer with live installer

lucasdemarchi, 2017-05-16:
This allows us to use network while handling the installation.

avinash-palleti, 2017-05-30:
I added new recipe file inside recipes-core/images/ as intel-aero-image-initramfs.bb and updated local.conf file
https://gist.github.com/avinash-palleti/cdad8c61370309019d8be72d57591f78
i am getting cpio.gz inside deploy/images/intel-aero/ but it is not getting bundled with iso .. 
iso still having initrd and kernel command line is still initrd = /initrd

@whbruce @bluelightning Any suggestions?

avinash-palleti, 2017-05-30:
We added new recipe file inside recipes-core/images/ as intel-aero-image-initramfs.bb and updated local.conf file
https://gist.github.com/avinash-palleti/cdad8c61370309019d8be72d57591f78

We also added INITRAMFS_IMAGE = "intel-aero-image-initramfs" inside machine.conf

we are getting cpio.gz inside deploy/images/intel-aero/ but it is not getting bundled with iso ..
iso still having initrd and kernel command line is still initrd = /initrd

Do we need to set anything for CONFIG_INITRAMFS_SOURCE kernel variable ?

@whbruce @bluelightning Any suggestions?

bartosh, 2017-06-08:
Hi, 

I'd like to help with this, but need a bit more information.

Can you provide more details about your setup (machine, layers used, etc) and about the goal. So far I understood that you need an installer image with enabled networking.

Is there any chance to debug this in qemu or I must have a hardware to reproduce your setup?

Regards,
Ed

avinash-palleti, 2017-06-08:
@bartosh Thank you :)

Goal is to migrate from initrd to initramfs live installer. 
We are working on Intel-aero board which is CHT based. 

AFAIK with initramfs live installer there should be cpio.gz file which goes into ramfs, ISO will include that cpio.gz and kernel command line will change respectively. 

When i do above mentioned changes, i can see cpio.gz is getting generated but that is not getting bundled with ISO.
Please tell me IRC name so that i will give more details there.

bartosh, 2017-06-08:
First of all there is an old confusion in terms of initramfs and initrd. initrd is a file system, which is mounted by kernel, but initramfs is a cpio archive. Yocto uses initramfs. The fact that it's called initrd doesn't mean it's real initrd. It's still initramfs.

Looking at the code in populate_kernel function I'd say that your initramfs was probably added to another one - microcode.cpio. Kernel should be able to unpack this combined archive.

Please, check size of initrd file in your image. If it's much bigger than tmp/deploy/images/intel-aero/microcode.cpio than you have your initramfs archive added to it.

I didn't use your patch yet. I've just added INITRD_IMAGE_LIVE = "core-image-minimal-initramfs" to conf/local.conf, built intel-aero-image and checked initrd size and size of core-image-minimal-initramfs and microcode.cpio:

$ sudo mount -o loop ./tmp/deploy/images/intel-aero/intel-aero-image-intel-aero.iso /mnt
mount: /dev/loop0 is write-protected, mounting read-only
$ ls -lh /mnt/initrd 
-r--r--r-- 1 root root 8.5M Jun  8 17:11 /mnt/initrd

$ ls -lh ./tmp/deploy/images/intel-aero/core-image-minimal-initramfs-intel-aero-20170608135642.rootfs.cpio.gz
-rw-r--r-- 2 ed users 7.6M Jun  8 16:58 ./tmp/deploy/images/intel-aero/core-image-minimal-initramfs-intel-aero-20170608135642.rootfs.cpio.gz

$ ls -lh ./tmp/deploy/images/intel-aero/microcode_20170511.cpio 
-rwxr-xr-x 2 ed users 965K Jun  8 14:12 ./tmp/deploy/images/intel-aero/microcode_20170511.cpio

As you can see initrd size is a sum of 2 component sizes.

Does this make sense to you?

lucasdemarchi, 2017-06-08:
> Goal is to migrate from initrd to initramfs live installer.

Goal is actually to boot a full image. In other words, the installer runs on a fully booted image, not a "boot to initrd type of image we have today".  What we want to do is to be able to have all service fully running and being able to install the image to the final location - this way we have network up and running, mavlink-router up and running and possibly an http server up and running to guide the installation.

bartosh, 2017-06-08:
Then I'd suggest to look at wic image type. There are at least 3 reasons for this:

1. The approach if much cleaner - instead of using initramfs it boots full image.
2. live image type is going to be deprecated sooner or later and replaced by wic
3. I'm a maintainer of wic, so can help :)

whbruce, 2017-06-08:
+1 on moving to wic.

avinash-palleti, 2017-06-08:
@bartosh 
Ya it makes sense now.. I was actually expecting cpio.gz inside ISO thats why got confused seeing still initrd inside. 
Now as lucas is expecting if we need full image to boot with initramfs we need to have "PACKAGE_INSTALL += INSTALL_IMAGE" which will make all packages as part of initramfs archive.

Compared to this way, moving to wic will be cleaner as you suggested.

bartosh, 2017-06-08:
I'd like to point out that initramfs uses device memory. the bigger the image is the more memory is used. This is one more reason to move to wic.

Anyway, let me know if you need further help with this.

lucasdemarchi, 2017-06-08:
Is there an equivalent of the initram installer on the wic image?

bartosh, 2017-06-08:
@lucasdemarchi good question. Here you can see how to add install target to boot loader: https://bugzilla.yoctoproject.org/show_bug.cgi?id=11044

Frankly I don't know if it's enough. I need to look how this functionality is implemented. Hopefully it won't be hard to implement it. As far as I understood you'd need to modify default installer anyway, so it shouldn't stop you I guess.

lucasdemarchi, 2017-06-08:
Afair this will just instruct it to stop at the initrd in order to run the installer. What we need is a new installer that runs on the live boot

bartosh, 2017-06-08:
Absolutely. So, we have 2 alternatives:
bring all you need (network up and running, mavlink-router up and running and possibly an http server up and running etc) into initramfs and change the installer
or
bring the installer into full image that already has(or can easily have) all of the above.

It's your choice :)

lucasdemarchi, 2017-06-08:
> bring the installer into full image that already has(or can easily have) all of the above.

This one, of course.

bartosh, 2017-06-08:
OK, here is the current install script: https://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/recipes-core/initrdscripts/files/init-install-efi.sh. 7 K of shell code. It copies rootfs from rootfs image to the target root partition. You should decide if you want the same approach or you can somehow use existing root partition or create another partition just for this purpose.

bartosh, 2017-06-09:
I'd suggest you to have a look at how this is implemented in RefKit:
- https://github.com/intel/intel-iot-refkit/blob/master/meta-refkit-core/classes/image-installer.bbclass
- https://github.com/intel/intel-iot-refkit/blob/master/meta-refkit-core/recipes-images/images/refkit-installer-image.bb