intel-aero/meta-intel-aero #150: Replace custom spidev modules

lucasdemarchi, 2017-05-16:
Instead of using custom modules to just export spidev, we should rather adopt an ACPI overlay. Currently we have 3 modules:

```
[    4.197174] spidev CAN registered
[    4.198505] spidev fpga registered
[    4.200892] spidev-imu: spidev registered
```

An example for similar ACPI override can be found at https://github.com/westeri/meta-acpi/blob/master/recipes-bsp/acpi-tables/samples/minnowboard-max/spidev.asl

avinash-palleti, 2017-05-17:
@lucasdemarchi I will look into this

avinash-palleti, 2017-05-26:
@lucasdemarchi https://lists.yoctoproject.org/pipermail/linux-yocto/2016-February/004604.html

Is this patch required to be under linux-yocto ?

avinash-palleti, 2017-05-26:
@lucasdemarchi I made changes as per sample in meta-acpi layer then realized format is little different after looking into intel-aero dsdt acpi table.

Below is my updated patch: https://github.com/avinash-palleti/meta-intel-aero-base/commit/ebe9d15297ead78b61e0ec06bbdda5c2e897d053

Still i am not seeing /dev/spidev*. As you mentioned CAN controller is already there as part of DSDT table but even that is also not seen in device. 
I referred below link and added a patch to linux, still no luck. 
http://gumstix.8.x6.nabble.com/Unable-to-get-dev-spidev-on-yocto-1-8-td4970184.html
linux Patch: https://gist.github.com/avinash-palleti/273fda6e5aafc3ed664b22d6942e1d28
With this patch all the existing acpi tables also not seen with acpidump command.

@whbruce Do you have any suggestion ?

whbruce, 2017-05-26:
Sorry, don't know about this feature, but it seems the right way to go. Maybe @clsulliv can help.

whbruce, 2017-06-20:
Still on my list. This should be an intel-aero-base issue.