intel-aero/meta-intel-aero #151: Do not start syslog

lucasdemarchi, 2017-05-16:
We are already running the journal, there's no real need for running syslog. Edit journal configuration so it doesn't forward its log to syslog and ensure syslog is not in the final image.

avinash-palleti, 2017-05-19:
@rjasuja is working on this

Rjasuja, 2017-05-22:
@lucasdemarchi i made changes as per this https://pastebin.com/uatX0PhC. 
But the "CONFIG_FEATURE_SYSLOG=y" is still enabled, i cannot override this(hidden feature).
Is there any other way to try this ? 
i used menuconfig to generate the config file(fragment.cfg).

avinash-palleti, 2017-05-22:
changes copied into gist: https://gist.github.com/avinash-palleti/056616b0bbe50f8b27ec24ca2505b3df

lucasdemarchi, 2017-05-22:
> changes copied into gist: https://gist.github.com/avinash-palleti/
> 056616b0bbe50f8b27ec24ca2505b3df

busybox-syslog is a package. Doesn't it work if you just remove it?

Rjasuja, 2017-05-22:
@lucasdemarchi i could not find any package installed named busybox-syslog(checked using bitbake -s)

Rjasuja, 2017-05-24:
https://github.com/Rjasuja/meta-intel-aero/commit/95eb2e20a3412ba2dd0dd01f01ec2ff4c5aaa9dc
https://github.com/Rjasuja/meta-intel-aero/commit/777f476b09cc1528275cb5a0700145585340ae2b

These are the two changes i have done. Please review

closed 2017-05-26 by avinash-palleti