intel-aero/meta-intel-aero #152: remove nfsd

lucasdemarchi, 2017-05-16:
nsfsd is failing to start, but it should actually not being started... it needs to be removed from image.

avinash-palleti, 2017-05-18:
@lucasdemarchi nfs-utils are coming from 
poky/meta/recipes-extended/packagegroups/packagegroup-core-full-cmdline.bb

any other way to remove instead of making changes to above file.

avinash-palleti, 2017-05-18:
Oneway is to override same packagegroup in our layer.

lucasdemarchi, 2017-05-18:
I think there's a way to blacklist the package, avoiding it to be installed. Otherwise I'm ok leaving it as is, as long as we don't try to start it on boot. You can do this by adding a .bbappend and set it not to autoenable. Another option may be a .bbappend and remove it from the packagegroup. @hbruce, any suggestion?

avinash-palleti, 2017-05-18:
http://www.yoctoproject.org/docs/current/ref-manual/ref-manual.html#var-PACKAGE_EXCLUDE

using this we can do in image recipe or local.conf

lucasdemarchi, 2017-05-18:
since we are including it at the "image level", we can set this just after including packagegroup-core-full-cmdline, with a comment saying it comes from that group.

padmashree9110, 2017-05-18:
@lucasdemarchi Getting below issue if we remove using PACKAGE_EXCLUDE either in local.conf or in image bb file.

https://gist.github.com/anonymous/3a253fdf5701dd6147b81452725a5efa

sugnanprabhu, 2017-05-18:
@padmashree9110 

try adding the following line which will fix this error.
RDEPENDS_packagegroup-core-full-cmdline-sys-services_remove = "    nfs-utils"

padmashree9110, 2017-05-18:
@sugnanprabhu 
yes, Its working now.
Thank you

lucasdemarchi, 2017-05-18:
@padmashree9110 please open a PR for master-next branch with the change

whbruce, 2017-05-18:
@sugnanprabhu nice work. @padmashree9110  make sure PACKAGE_EXCLUDE and RDEPENDS_remove are in the same place so people know what you've done and why.

lucasdemarchi, 2017-05-18:
Taking a look now on what is each packagegroup.... Another approach would be to include packagegroup-core-full-cmdline-utils and packagegroup-core-full-cmdline-extended, but not the rest. Or everything except packagegroup-core-full-cmdline-sys-services

sugnanprabhu, 2017-05-19:
@lucasdemarchi @whbruce I also tried using BBMASK, but still couldn't avoid using RDEPENDS_remove 

http://www.yoctoproject.org/docs/current/ref-manual/ref-manual.html#var-BBMASK

lucasdemarchi, 2017-05-19:
What about reducing the package groups we install as suggested on my previous comment? I'm ok with using the RDEPENDS_remove too

avinash-palleti, 2017-05-19:
@lucasdemarchi I will review package groups that are being installed and try to come up with minimal list. For now @padmashree9110 will submit PR to fix this issue.

padmashree9110, 2017-05-26:
@lucasdemarchi If we remove packagegroup-core-full-cmdline-sys-services as you suggested, we are going to remove below utilities from image,

at \
    bzip2 \
    cronie \
    dbus \
    dbus-glib \
    elfutils \
    gzip \
    less \
    libcap \
    libevent \
    logrotate \
    nfs-utils \
    pciutils \
    libpcre \
    rpcbind \
    sysfsutils \
    tcp-wrappers \
    tzdata \

Few utilities seems to important "gzip sysfsutils etc." Do we have these utils somewhere else as well.

closed 2017-05-30 by lucasdemarchi

adityavarmatcs, 2017-05-31:
On additional note, I see most of the listed utilities of packagegroup-core-full-cmdline-sys-services neither working and nor available /bin directory except below. 
gzip
less
rpcbind