intel-aero/meta-intel-aero #154: Test usb-ethernet on windows

lucasdemarchi, 2017-05-18:
Our ethernet over USB is very reliable for me now on master-next. Aero is configured to give an IP in the 192.168.7.10 - 192.168.7.30. This needs to be tested on Windows and instructions written (can be on this issue) of what needs to be done. Just getting an IP is sufficient.

lucasdemarchi, 2017-06-16:
It worked for me on windows 8 after installing cdc drivers for BBB. Strangely it continued to work after I uninstalled those. Windows 10 without additional drivers... it didn't work. @guermonprez can you help here?

guermonprez, 2017-06-16:
I don't have a "naive" windows install at my disposal: I installed tons of USB-ethernet drivers over the years so it wouldn't be a good test.
For the default install procedure, I do not show this usb-ethernet method as it's hard to diagnose problems clients may face (ubuntu network manager, routing, ...). I'm using wifi to keep it simple. But it would be a good thing to have documented as a reference on the github doc.

lucasdemarchi, 2017-06-23:
> For the default install procedure, I do not show this usb-ethernet method as it's hard to diagnose problems clients

usb-ethernet is working fine on Linux and Mac, including the local name resolution. I don't expect local name resolution to be working now for windows, but at least it should get an IP and be able to connect. That's what I was asking if you could test. If you can't that's fine.