intel-aero/meta-intel-aero #159: 32 bit support

ohshout, 2017-05-25:
Hi friends,

Sorry if I'm posting at the wrong place. The intel-aero forum might not be the best place for this kind of questions I guess.. My question is:

Whether it is possible to build everything (including the kernel) to be 32-bit x86? I'm not a yocto/bitbake expert, but I'm ok with tinkering with the yocto configurations. Just want to make sure there isn't any significant obstacles stopping me from achieving that.

Thanks a lot!
Tom.

lucasdemarchi, 2017-05-25:
It is possible, just can't imagine why you would want that.... But anyway what you will need to do is to change the machine configuration to inherit from intel-core2-32 rather than intel-core-i7-64. See https://github.com/intel-aero/meta-intel-aero-base/blob/master/conf/machine/intel-aero.conf#L7

You may rather want to add multilib, which would allow you to execute 32 bit binaries.

ohshout, 2017-05-25:
Thanks for the quick reply, Lucas! We're running an old hypervisor on the aero board that only supports 32 bit guests... It is pain in the neck... Anyway,  I'll try it out now. Thanks again.

Tom.

ohshout, 2017-06-05:
Hi Lucas,

Sorry for bothering you again... I just tried the above solution out, it gives me the following error. I'm gonna look into it today, but if you can give me some hints, it will be highly appreciated.

Thanks,
Tom.

_[czq@mrbig build]$ bitbake intel-aero-image
Loading cache: 100% |#############################################################################| ETA:  00:00:00
Loaded 2778 entries from dependency cache.
Parsing recipes: 100% |###########################################################################| Time: 00:00:00
Parsing of 2215 .bb files complete (2209 cached, 6 parsed). 2783 targets, 109 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'gma500-gfx-check' (but /scratch/intel_aero/poky/meta/recipes-core/packagegroups/packagegroup-core-boot.bb RDEPENDS on or otherwise requires it)
ERROR: gma500-gfx-check was skipped: incompatible with machine intel-aero (not in COMPATIBLE_MACHINE)
ERROR: gma500-gfx-check was skipped: incompatible with machine intel-aero (not in COMPATIBLE_MACHINE)
NOTE: Runtime target 'gma500-gfx-check' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['gma500-gfx-check']
NOTE: Runtime target 'packagegroup-core-boot' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-boot', 'gma500-gfx-check']
ERROR: Required build target 'intel-aero-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['intel-aero-image', 'packagegroup-core-boot', 'gma500-gfx-check']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
_

lucasdemarchi, 2017-06-05:
gma500-gfx-check is not necessary in our board. Try removing it from packagegroup-core-boot. See our intel-aero-image recipe on how we remove some packages.

ohshout, 2017-06-06:
Lucas,

I removed the line:
MACHINE_ESSENTIAL_EXTRA_RRECOMMENDS += "gma500-gfx-check"
from the file meta-intel/conf/machine/intel-core2-32.conf

Then I did a cleanall, then build, it's not complaining about gma500 any more, but I'm loads of errors, all looking similar:
ERROR: linux-yocto-4.4.60+gitAUTOINC+2cc78e92f4-r0 do_package_qa: QA Issue: Architecture did not match (3 to 62) on work/core2-32-intel-common-poky-linux/linux-yocto/4.4.60+gitAUTOINC+2cc78e92f4-r0/packages-split/kernel-module-cast6-generic/lib/modules/4.4.60-yocto-standard/kernel/crypto/cast6_generic.ko [arch]

Again, any quick hints?

Thanks!

lucasdemarchi, 2017-06-07:
Edit the kernel config to switch to 32 bits.

Again... I would reconsider why your are doing this in the first place. Either make your binary 64bits or... we can run userspace 32 bit binaries just fine when multilib is enabled.

closed 2017-06-13 by ohshout