intel-aero/meta-intel-aero #165: Disable/change screensaver and energy saving.

hzhutanx, 2017-06-09:
The Aero RTF drone has a 10 minute input timeout for inactivity. When this occurs, the HDMI output display will blank and the monitor will turn off. This occurs even when running the OS bundled app cpp-capture to display some camera features. It's desired to have an option to allow the user to disable this.

guermonprez, 2017-06-09:
agree
I think the 3 lines will do the trick:
xset -dpms
xset s off
xset s noblank

hzhutanx, 2017-06-09:
This would be particularly useful for users who want a set-and-forget method that sticks across drone restarts.

avinash-palleti, 2017-06-12:
@hzhutanx Those 3 commands which @guermonprez mentioned are working... Are you looking for this to persist across reboot ?

anselmolsm, 2017-06-12:
These configs can be added to either xorg.conf or to a file in xorg.conf.d.
@avinash-palleti , please take a look at the following commit, it might be something like this:
https://github.com/anselmolsm/meta-intel-aero-base/commit/a9dd94f0a9c833070c1decda68d42bd17271452f

hzhutanx, 2017-06-12:
Yes, across a reboot.
I've particularly used the following command withing my terminal session.
xset s off -dpms

but whatever works best.

avinash-palleti, 2017-06-13:
thanks @anselmolsm . Its working, with this end user can update required time out and this persists across reboots.

https://github.com/intel-aero/meta-intel-aero-base/pull/6

closed 2017-06-23 by lucasdemarchi