intel-aero/meta-intel-aero #168: Issue with patch that was submitted for #152

adityavarmatcs, 2017-06-13:
While building QEMU Emulator, below error was thrown.

Code was synced on:    June 6, 2017
QEMU Emulator build procedure:

repo init -u https://github.com/intel-aero/intel-aero-manifest.git -b master

repo sync

# Initialize build environment variables by running

export TEMPLATECONF=../meta-intel-aero/conf

source poky/oe-init-build-env

In local.conf file added MACHINE ?= "qemux86-64" to build image for qemu emulator

# Build intel-aero-image

bitbake intel-aero-image

----------------------------------------------------------------------------------------------------------------------------------

Error: package packagegroup-distro-base-1.0-r83.qemux86_64 requires packagegroup-core-device-devel, but none of the providers can be installed
  - package packagegroup-base-1.0-r83.qemux86_64 requires packagegroup-distro-base, but none of the providers can be installed
  - package packagegroup-core-device-devel-1.0-r1.noarch requires nfs-export-root, but none of the providers can be installed
  - package packagegroup-base-extended-1.0-r83.qemux86_64 requires packagegroup-base, but none of the providers can be installed
  - package nfs-export-root-1.0-r1.core2_64 requires packagegroup-core-nfs-server, but none of the providers can be installed
  - conflicting requests
 

- nothing provides nfs-utils needed by packagegroup-core-nfs-server-1.0-r2.noarch
(try to add '--allowerasing' to command line to replace conflicting packages)

ERROR: intel-aero-image-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/ubuntu/intel_aero/build/tmp/work/qemux86_64-poky-linux/intel-aero-image/1.0-r0/temp/log.do_rootfs.24278
ERROR: Task (/home/ubuntu/intel_aero/poky/../meta-intel-aero/recipes-core/images/intel-aero-image.bb:do_rootfs) failed with exit code '1'

----------------------------------------------------------------------------------------------------------------------------------
Workaround done:
Image got built when the below two lines are commented in /intel_aero/build/conf/local.conf

Cause:
According to the patch that was submitted for #152(https://github.com/intel-aero/meta-intel-aero/issues/152), nfs-utils package is removed from the target image.

+PACKAGE_EXCLUDE="nfs-utils"
+RDEPENDS_packagegroup-core-full-cmdline-sys-services_remove=" nfs-utils"

However other package ("packagegroup-core-nfs-server") has dependency on nfs-utils - Hence error thrown.

Note:
packagegroup-core-nfs-server is  included in  /home/ubuntu/intel_aero/poky/meta/recipes-devtools/devel-config/nfs-export-root.bb file

We might have to introduce RDEPENDS_remove value for packagegroup-core-nfs-server to remove the dependency with nfs-utils

----------------------------------------------------------------------------------------------------------------------------------

anselmolsm, 2017-06-14:
Hi @adityavarmatcs, I successfully built the image after a repo sync today. I was expecting to get this error, but it did not happen. Could you please give it another try, after a new repo sync? Thanks.

adityavarmatcs, 2017-06-15:
This error was encountered while building the image for QEMU Emulator (config as ) - However I shall try syncing again and check if those errors are still valid.
Thanks.

anselmolsm, 2017-06-15:
I'll rebuild targeting QEMU. If the same repo state fails in that particular case, there might be something else to be fixed then.

adityavarmatcs, 2017-06-28:
I am trying to build a code of MASTER branch as I need "Systemd" to be up and running. Is it recommended to flash MASTER image (built one) on the board? Is MASTER (until now) stable? Can you advise?

anitha-suresh, 2017-07-03:
Yes, you can flash the MASTER image on the board.

adityavarmatcs, 2017-07-04:
Successfully flashed master image on the board.

Thank you

closed 2017-07-04 by adityavarmatcs