intel-aero/meta-intel-aero #169: How to use ttyS0

ohshout, 2017-06-14:
Good Morning!

Just a quick question, how do you use the ttyS0 on the aero board. 
According to this doc: https://communities.intel.com/servlet/JiveServlet/download/453901-164636/Intel_Aero_Compute_Board_Hardware_Feature_Usage_v1.pdf, It seems to me the ttyS0
is the Pin 9 and 10 on J1 in Section 3.8.1. But I couldn't get any output from it.

Am I missing any setup?
Thanks,
Tom.

rohithnlr, 2017-06-15:
If you want to see the console logs check the /etc/inittab file
and check whether there is a line like this "s0:12345:respawn:/bin/start_getty 115200 ttyS0"
if not add this.

avinash-palleti, 2017-06-15:
If it's the board with aerofc connected, you can simply disconnect it.
Then edit the kernel command line on the bootloader and replace ttyS0 with ttyS1. This should allow you to use the serial pins on the power connector to see the logs. You need TTL to USB cable to connect.

ohshout, 2017-06-15:
Thanks for the replies!

I checked inittab, the line is there. I use a tty to USB serial cable and run minicom on my desktop. This setup gives me output when hooked up to ttyS1(the pins with the power cable). But weirdly I got nothing when hooked up to ttyS0(the pins in my original post), not even logs when the kernel boots. 

So I suspect I'm using the wrong pins. Can anyone confirm the pins I'm using?

Btw, I actually got the fc connected and that's why I want to use ttyS0 to see the logs.

Thanks,
Tom.

lucasdemarchi, 2017-06-15:
ttyS0 is not connected there. You will need to use an USB serial to get an additional uart.

That uart is meant for the flight controller and on the RTF it goes to the ESCs.

ohshout, 2017-06-15:
@lucasdemarchi I think you misunderstood what I said:) I meant the ttyS1 on my board has already been taken, that's why I have to use ttyS0 as a console (or at least to see the kernel boot log). Right now I hooked up a USB serial to ttyS0 and got no output.

lucasdemarchi, 2017-06-15:
No, I didn't misunderstand. ttyS0 from the SoC is not connected there. It would be awesome to have it, but we don't.

SW alternative that I can think of would be to have a sdio<->uart bridge implemented on the fpga. We have an sdio from SoC connected there. Then we would dedicate it to talk to the flight controller and free up ttyS1 for console. There are some IPs out there with that bridge, some that even have the kernel driver implemented.

ohshout, 2017-06-15:
Okay, I see. Thanks, Lucas! 
So to conclude, right now, there's no I/O port based (instead of memory mapped) UART can be used by the linux.

closed 2017-06-23 by lucasdemarchi