intel-aero/meta-intel-aero #178: Reevaluate eee-acpi-scripts

anselmolsm, 2017-06-22:
This issue is to remind us to continue https://github.com/intel-aero/meta-intel-aero-base/pull/9 in future.

padmashree9110, 2017-07-20:
@anselmolsm I have seen IRC chat and verified Power on/off with button before submitting PR.(forgot to add a note)
Is there anything else that I need to verify.

anselmolsm, 2017-07-26:
@lucasdemarchi raised the concern that the soft shutdown is not working properly in 1.4.
@avinash-palleti, could you please share your findings about it?  Thanks.

anselmolsm, 2017-07-27:
Thanks @avinash-palleti . With regards to the evaluation, I think it's done. If we identify there's a problem with the shutdown, we'll report a new issue to track that as a bug.

closed 2017-07-27 by anselmolsm