intel-aero/meta-intel-aero #185: sample app capture_example binary not found

lbegani, 2017-07-04:
To test the cameras in intel-aero using sample app, I tried running the following command but without success - 
root@intel-aero:~# C=10 INPUT=0 MODE=PREVIEW capture_example --userp -d /dev/video13
It seems capture_example bin is not part of the image. The github project for sample-apps "git://github.com/intel-aero/sample-apps.git" is included in spi-xfer.bb file. This downloads the sources from sample-apps/capturev4l2 to <>/spi-xfer/*. However there is no command to compile and install the capture sample app.
1. Should we include the compile/install commands for capture app in spi-xfer ?
2. Or should we add a new bb file specifically for capture command (& remove the dup code under spi-xfer)?

lkm1321, 2017-07-04:
I think this was a legacy app to demonstrate some functionality of the OVT cameras. The OVT cameras seem barely functional as mentioned in prev issues. What I did was I just scp'ed the source files from my host machine and built it on the drone itself. Does what it's supposed to do for the forward OVT camera.

lbegani, 2017-07-06:
My concern was whether capture sample app needs to be included by default in the builds for the end users.

anselmolsm, 2017-07-11:
We could have a single sample-apps.bb that generates packages for each sample. We need to revisit those samples and check which one still make sense. Those packages could be offered in the rpm repo.

closed 2017-07-12 by lucasdemarchi