intel-aero/meta-intel-aero #189: wifi failing authentication right after boot

lucasdemarchi, 2017-07-06:
#93 strikes again.

rng-tools is not being installed anymore. This or an equivalent package needs to run in order to provide enough entropy for wpa_supplicant to authenticate clients right after boot.

Log from wpa_supplicant:
```
May 30 19:45:58 intel-aero wpa_supplicant[420]: WPS: Converting push_button to virtual_push_button for WPS 2.0 compliance
May 30 19:45:58 intel-aero wpa_supplicant[420]: WPS: Converting push_button to virtual_push_button for WPS 2.0 compliance
May 30 19:45:59 intel-aero wpa_supplicant[420]: nl80211: deinit ifname=wlp1s0 disabled_11b_rates=0
May 30 19:45:59 intel-aero wpa_supplicant[420]: WPS: Converting push_button to virtual_push_button for WPS 2.0 compliance
May 30 19:45:59 intel-aero wpa_supplicant[420]: Note: nl80211 driver interface is not designed to be used with ap_scan=2; this can result in connection failures
May 30 19:45:59 intel-aero wpa_supplicant[420]: WPS: Converting push_button to virtual_push_button for WPS 2.0 compliance
May 30 19:45:59 intel-aero wpa_supplicant[420]: Using interface wlp1s0 with hwaddr 00:c2:c6:d3:0a:22 and ssid "Aero-00C2C6D30A22"
May 30 19:45:59 intel-aero wpa_supplicant[420]: WPS: Converting push_button to virtual_push_button for WPS 2.0 compliance
May 30 19:45:59 intel-aero wpa_supplicant[420]: random: Cannot read from /dev/random: Resource temporarily unavailable
May 30 19:45:59 intel-aero wpa_supplicant[420]: random: Only 0/20 bytes of strong random data available from /dev/random
May 30 19:45:59 intel-aero wpa_supplicant[420]: random: Not enough entropy pool available for secure operations
May 30 19:45:59 intel-aero wpa_supplicant[420]: WPA: Not enough entropy in random pool for secure operations - update keys later when the first station connects
May 30 19:45:59 intel-aero wpa_supplicant[420]: wlp1s0: interface state UNINITIALIZED->ENABLED
May 30 19:45:59 intel-aero wpa_supplicant[420]: wlp1s0: AP-ENABLED
May 30 19:45:59 intel-aero wpa_supplicant[420]: wlp1s0: CTRL-EVENT-CONNECTED - Connection to 00:c2:c6:d3:0a:22 completed [id=0 id_str=]
May 30 19:46:10 intel-aero wpa_supplicant[420]: random: Cannot read from /dev/random: Resource temporarily unavailable
May 30 19:46:10 intel-aero wpa_supplicant[420]: random: Only 6/20 bytes of strong random data available from /dev/random
May 30 19:46:10 intel-aero wpa_supplicant[420]: random: Not enough entropy pool available for secure operations
May 30 19:46:10 intel-aero wpa_supplicant[420]: WPA: Not enough entropy in random pool to proceed - reject first 4-way handshake
May 30 19:46:14 intel-aero wpa_supplicant[420]: random: Cannot read from /dev/random: Resource temporarily unavailable
May 30 19:46:14 intel-aero wpa_supplicant[420]: random: Only 6/20 bytes of strong random data available from /dev/random
May 30 19:46:14 intel-aero wpa_supplicant[420]: random: Not enough entropy pool available for secure operations
May 30 19:46:14 intel-aero wpa_supplicant[420]: WPA: Not enough entropy in random pool to proceed - reject first 4-way handshake
May 30 19:46:15 intel-aero wpa_supplicant[420]: random: Cannot read from /dev/random: Resource temporarily unavailable
May 30 19:46:15 intel-aero wpa_supplicant[420]: random: Only 6/20 bytes of strong random data available from /dev/random
May 30 19:46:15 intel-aero wpa_supplicant[420]: random: Allow operation to proceed based on internal entropy
May 30 19:46:15 intel-aero wpa_supplicant[420]: wlp1s0: AP-STA-CONNECTED 30:b5:c2:16:b0:52
```

anselmolsm, 2017-07-06:
Have you had the chance to identify why rng-tools pkg is not being installed anymore? It's still in the intel-aero-image.bb recipe.

anselmolsm, 2017-07-06:
/usr/sbin/rngd is in the latest image. However it returns "read error" when executed.

lucasdemarchi, 2017-07-07:
Not sure... Maybe related to the order it is started. Need to check what is the hw device it's trying to use. Arch's page have more details... If nothing works we can fallback to haveged for the 1.4 release.

zehortigoza, 2017-07-07:
https://github.com/intel-aero/meta-intel-aero/pull/192 fixes this

closed 2017-07-09 by anselmolsm