intel-aero/meta-intel-aero #195: AP mode on/off

anselmolsm, 2017-07-09:
For the record and further evaluation. A suggestion we received about turning AP mode on/off:

Would it be possible to provide a way to make the connman AP mode state either persistent or configurable using the physical button (e.g. with a triple press)?

lucasdemarchi, 2017-07-09:
It is possible, not sure if desired. When you press the button there's an event generated that currently is captured to turn the board on/off.

I rather think connecting the cable and changing the mode may be better... Not sure. Particularly when cockpit is ready and people can use the web interface.

padmashree9110, 2017-07-19:
@lucasdemarchi @anselmolsm Shall we have this option as part of Network plugin only ?
https://github.com/intel-aero/meta-intel-aero/issues/200

lucasdemarchi, 2017-07-19:
Yes

Rjasuja, 2017-07-27:
@lucasdemarchi Following is the snapshot, i am planning to implement for this.
[image]

Function: 
1. On the press of Scan button, it will list all available AP's in text section as a list. From the list, whatever will be selected, will be used to connect. In the back end, a python script will be executed on pressing Scan button. This script will list down AP available in a file which javascript will use to display in the List menu.
2. Selected AP from the list and password entered in the input box will be used to connect. This will also invoke a python script to connect.

Using javascript, i could not get a way to execute interactive shell commands. With Python, it is easy to execute such tasks.
Any suggestions ?

Rjasuja, 2017-07-28:
@lucasdemarchi  Proceeding with Python, i'll need pexpect for executing commands with connmanctl shell. Should we include this as part of image ?

lucasdemarchi, 2017-07-28:
If it's necessary, yes. I think we have Python's subprocess though.

Rjasuja, 2017-07-28:
Subprocess doesn't help with interactive shell.

anselmolsm, 2017-07-28:
You don't need the interactive shell. `connmanctl` also accepts commands as arguments.
`connmanctl scan wifi`, `connmanctl services`, etc.

Rjasuja, 2017-07-31:
For switching to Station Mode, it needs interactive shell and it doesn't connect without an interactive shell. I tried a few times(giving as an argument) but it gives : "Not registered ". Although, it got connected with the shell.

Rjasuja, 2017-08-01:
https://github.com/Rjasuja/aero-cockpit-plugins/blob/upstream8/network/network.html
I am facing an issue with opening a file and reading its contents using javascript.
Issue is : Javascript(using XMLHttpRequest) will read the same file(first read) over and again even though the file contents have changed.
Tried the following : 
1. If i change the file name in the second attempt, i get different content.
2. If i take a reference of same file in new variable(second attempt), still data doesn't change.

Any suggestions ?

shakthi-prashanth-m, 2017-08-01:
As the python script writes scan results to a local file, can you try reading using normal file read operation ?

Rjasuja, 2017-08-02:
Created a PR : https://github.com/intel-aero/aero-cockpit-plugins/pull/8
#201 (Added : Client connection configuration: scan, select and connect to a WiFi network)