intel-aero/meta-intel-aero #199: cockpit plugin: about

anselmolsm, 2017-07-11:
Add a simple cockpit plugin to show info about Intel Aero.
  - OS
  - FPGA
  - FC
  - BIOS

This could get the output of aero-get-version.py.

avinash-palleti, 2017-07-12:
@anselmolsm Every plugin will have respective folder with set of files including manifest.json and html file.
lets have all plugin folders in /recipes-cockpit/plugins/, cockpit.bb will have installation commands to place them in target image at /usr/share/cockpit/ location.

Any other plan ?

anselmolsm, 2017-07-13:
@avinash-palleti We might change the way the current .bb installs cockpit components. Also, I believe plugins could be installed as separate packages as well. #206

anselmolsm, 2017-07-13:
BTW, this plugins has been submitted as https://github.com/intel-aero/aero-cockpit-plugins/pull/1

avinash-palleti, 2017-07-17:
PR Merged.

closed 2017-07-17 by avinash-palleti