intel-aero/meta-intel-aero #200: cockpit plugin: Network

anselmolsm, 2017-07-11:
Add a network plugin that interacts with the network management software (connman):
 - [ ] 1)  AP setup: Change SSID and password
 - [ ] 2)  Factory reset - restores the default wifi settings (SSID and password)

Rjasuja, 2017-07-14:
Submitted PR: https://github.com/intel-aero/aero-cockpit-plugins/pull/2

Rjasuja, 2017-07-14:
Attaching sample screenshots for the plugin:

[image]
[image]
[image]

avinash-palleti, 2017-07-26:
PR merged https://github.com/intel-aero/aero-cockpit-plugins/pull/5

closed 2017-07-26 by avinash-palleti