intel-aero/meta-intel-aero #201: cockpit plugin: Network part 2

anselmolsm, 2017-07-11:
Extend the network plugin, adding:
 - [ ] Client connection configuration: scan, select and connect to a WiFi network
 - [ ] USB route configuration

diksha-gupta, 2017-08-01:
Please find attached the screen-shots for your reference.
It lists down the available WiFiAP connections, and upon clicking it connects to the selected WiFi AP.
The back-end script is implemented using connmanctl tool which works for open WiFi AP successfully. Connecting to secured WiFi AP is still not successful yet - Working on it.
Please clarify more about USB route configuration.
[image]
[image]

Rjasuja, 2017-08-29:
Default NetworkManager plugin available with cockpit is to monitor the network. Attaching a screenshot : 
[image]

For switching modes in Wifi and connecting to a AP, an additional update is required.