intel-aero/meta-intel-aero #202: cockpit plugin: Software Update

anselmolsm, 2017-07-11:
Plugin that allows:

- [ ] system upgrade (flash new .iso):
  - USB drive:
     - Flash a .iso provided by the user and triggers aero-reboot-update.py
     - Check if version is available online
     - Download iso and prepare for update
- [ ] System update (dnf update) - evaluate if the existing cockpit plugin can be used
- [ ] FPGA update
- [ ] BIOS update

Rjasuja, 2017-07-14:
I am working on this issue

anselmolsm, 2017-07-14:
@Rjasuja I split this in 2 parts. See #216 for v1.5.

diksha-gupta, 2017-07-19:
As part of S/W update-plugin, is it required to consider AeroFC firmware update as well? It can be done by running aerofc_update.sh script present in /usr/sbin directory, with PX4 firmware file (copied in an agreed file path) as an argument.

anselmolsm, 2017-07-19:
I think I left AeroFC update out of the description by mistake. 

@diksha-gupta /etc/aerofc contains both PX4 and ArduPilot firwares for AeroFC. At first those files should be updated using rpm, not sure if we should allow firmware uploads in this plugin. @lucasdemarchi , opinions?

lucasdemarchi, 2017-07-19:
aerofc-update.sh accepts the firmware as an argument.

The plugin should:
a) list the available "factory firmwares" that can be get from /etc/aerofc
b) let the user to upload its own firmware: this should by no means write to /etc... the plugin should just accept the file, save it to a temporary location, call aerofc-update.sh and remove the file.

After doing that we let the plugin have a well documented http API so it's possible to interact with it via GCS.

diksha-gupta, 2017-07-24:
Here is the snapshot of s/w plugin development. 
1. BIOS update - Assume that capsule*.rpm is present in SD card or external media storage disk 
	Upon clicking on this button, BIOS UPDATE takes place and system reboots.
2. .jam update - The existing *.jam files (/etc/fpga) are listed here. Upon clicking on it, JAM update  takes place.
	Should we consider custom or updated *.jam file as well?
3. FC update - The exisitng *.px4 files and the ones placed in the external storage are listed here.   Upon clicking them, the respective update takes place.

This is how we are framing this use case.
Kindly evaluate the UI look and feel and let us know your comments.
[image]
[image]
[image]

avinash-palleti, 2017-07-24:
@diksha-gupta 
It will look nice if you can have tab structure. Please have look at "Services" plugin, it has multiple tabs each serves different functionality.
It will look good if you can list the existing jam files and FC Firmware in dropdown box instead of buttons.
There should be a way for user to update with custom file and also it should take latest release version from https://downloadcenter.intel.com/download/26748/Intel-Aero-Platform-for-UAVs-Installation-Files

lucasdemarchi, 2017-07-26:
I agree with @avinash-palleti. Just take care when checking online for updates since most of the times the board won't be connected to internet. You can use the client side with js , but even then care must be taken to handle fails gracefully.

vidhyagithub, 2017-07-26:
@avinash-palleti @lucasdemarchi 
Thanks for the comments.
We have modified the UI look and feel based on your suggestions. Kindly refer to the attached screen-shots and let us know your opinion on the same. If any further improvements is required, we shall work on them and submit PR.

Please clarify the below queries:
1. From the download centre (https://downloadcenter.intel.com/download/26748/Intel-Aero-Platform-for-UAVs-Installation-Files), we see that the latest BIOS version is now located @ https://downloadmirror.intel.com/26500/eng/capsule-01.00.12-r0.core2_64.rpm path which is of latest version i.e. 1.4 - In future, if any new BIOS VERSION is released, then the above URL may not be valid and we would have to again update our html file to point to the latest URL LINK - Is our understanding correct? Please comment.
2. We have assumed that custom BIOS and PX4 files shall always be placed at external hard disk. Hence, it is required for the user to copy the custom files in external hard disk only. Kindly let us know your view on this.
[image]
[image]
[image]
[image]

lucasdemarchi, 2017-07-26:
> From the download centre (https://downloadcenter.intel.com/download/26748/Intel-Aero-Platform-for-UAVs-Installation-Files), we see that the latest BIOS version is now located @ [...]

Yeah.... thinking again, currently there's no reliable way to determine the last released OS image and Bios.  Just remote that part from your code until that's available.

> We have assumed that custom BIOS and PX4 files shall always be placed at external hard disk. Hence, it is required for the user to copy the custom files in external hard disk only. Kindly let us know your view on this.

No, that's wrong.

For the UI, as @avinash-palleti said previously, please use the same way it's done on "Services"  plugin. Inspecting their code, it's this snippet here:

``` html
<div class="content-header-extra">
   <div class="btn-group" id="services-filter">
     <button class="btn btn-default" data-pattern="\.target$" translatable="yes">Targets</button>
     <button class="btn btn-default active" data-pattern="\.service$" translatable="yes">System Services</button>
     <button class="btn btn-default" data-pattern="\.socket$" translatable="yes">Sockets</button>
     <button class="btn btn-default" data-pattern="\.timer$" translatable="yes">Timers</button>
     <button class="btn btn-default" data-pattern="\.path$" translatable="yes">Paths</button>
   </div>
   <button id="create-timer" data-toggle="#timer-dialog" data-target="#timer-dialog" class="btn btn-primary pull-right" translate>Create Timer</button>
...
</div>
```

I used Chrome to inspect the source (F12), but you can also take a look on their source code as well.
You will also need a similar .css.

lucasdemarchi, 2017-07-26:
Their source:
https://github.com/cockpit-project/cockpit/blob/2954a22d01ef1305c8e2f42f84435f9140fe8d2e/pkg/systemd/services.html#L50-L59