intel-aero/meta-intel-aero #203: cockpit plugin: docker plugin

anselmolsm, 2017-07-11:
Evaluate the existing cockpit docker plugin and provide feedback, then a decision can be made if additional features are required.

Rjasuja, 2017-07-17:
Docker support in Cockpit includes : 
1. Getting New Images
2. Starting Docker server
3. Starting different containers(with different environment variables) from the same image
4. Manages Resources
5. Manages Container Resources.

Tried Downloading Ros container and launching roscore. It worked. Attaching screenshots.

Docker Statistics : 
[image]
Running Container: 
[image]
Resource Limits of docker : 
[image]
Roscore launched : 
[image]

lucasdemarchi, 2017-07-17:
Nice. Please send a PR enabling it back. Thanks.

Rjasuja, 2017-07-18:
Docker is enabled by default.

lucasdemarchi, 2017-07-18:
Oh, great! That means I didn't remove it... Checking https://github.com/intel-aero/meta-intel-aero/commit/73ebb0f49379b8f9c735f71362fc2a3861c3fe5a, indeed I didn't.

thanks for testing, it was good to have this nonetheless since we can use your tests to add image to the documentation later :)

closed 2017-07-18 by lucasdemarchi