intel-aero/meta-intel-aero #207: rng-tools: use hwrng

anselmolsm, 2017-07-12:
Configure rngd to use /dev/hwrng

avinash-palleti, 2017-07-12:
@anselmolsm i dont see device getting listed with lspci.
So insmod is failing for rng modules

avinash-palleti, 2017-07-12:
@anselmolsm i dont see device getting listed with lspci.
So insmod is failing for rng modules(intel-rng.ko, amd-rng.ko) with error "No such device"

But rngd command shows that there is hardware rng.

$ rngd -v 
read error
Available entropy sources:
      Intel/AMD hardware rng
      DRNG

when i get info of that device,

$ udevadm info /dev/hwrng
DEVPATH=/devices/virtual/misc/hw_random

anitha-suresh, 2017-07-12:
@anselmolsm, looks like our RNG is a CPU based one and hence it is not showing up in lspci?

avinash-palleti, 2017-07-12:
@anselmolsm I got confused because of read error, but it seems its ok to have couple of read error while starting the service.

I changed rngd.service file to use /dev/hwrng device as input device, after this change 
$ cat /dev/random 
gives bigger chunk of binary data, which means random number that is generated is more complex and actually used hw entropy.

I will submit the patch.

lucasdemarchi, 2017-07-12:
Also check if wifi (or SSH via usb-ethernet) doesn't give error in the first authentication right after boot

avinash-palleti, 2017-07-12:
@lucasdemarchi Will check it and update.

closed 2017-07-13 by lucasdemarchi