intel-aero/meta-intel-aero #208: cockpit plugin: SDK (updated) built as part of Yocto image

anitha-suresh, 2017-07-12:


anselmolsm, 2017-07-12:
This is a step after #202

lfelipe, 2017-07-12:
@anitha-suresh is this about providing information on updates to the Yocto SDK? Not sure whether cockpit would be a good location for that. Can you provide more description

anitha-suresh, 2017-07-13:
@lfelipe, I meant about the SDK which is built along with the Yocto image and hence I had mentioned that in the title (original title) . This would be beneficial to the app developers in getting latest SDK.

anitha-suresh, 2017-07-13:
As cockpit runs on the board, giving the option of Yocto SDK might not make much sense I guess. Maybe we need to figure out another way to help the app developer in sharing the link to update SDK and host it separately?

lfelipe, 2017-07-14:
Yes, that should likely just go through the standard download website and we can from cockpit just mention whats the latest release available at some point.

closed 2017-07-14 by lfelipe