intel-aero/meta-intel-aero #211: Add Aero branding

lucasdemarchi, 2017-07-12:
See branding directory in cockpit. Customize with logo, css, etc like other distributions do.

adityavarmatcs, 2017-08-01:
This can be achieved by,

1. Replacing the default logo file(logo.png) present in (/cockpit/branding/default) with "custom logo file"
2. /cockpit/branding/default/branding.css to include the following changes
    a. background-image of #badge to point to "custom logo file" 
    b. content of #brand to refer to "${PRETTY_NAME}" which will return the Name and Version of system distro

Is that fine to modify default folder only? or Do you want "custom folder" for brand info?
Can you clarify about the custom logo file?
Below are the screen-shots for your quick reference.
[image]

lucasdemarchi, 2017-08-01:
We should not modify the files in default folder. We have to do like the other distros do adding their own folder.