intel-aero/meta-intel-aero #213: Error in Mission Planning from QGC

SaliniVenate, 2017-07-13:
While uploading a mission with RTL option to Intel Aero from QGC I get the following error:
1) Error during Mission communication with Vehicle: Vehicle returned error: Command not supported (MAV_MISSION_UNSUPPORTED).Command(,,20)

[image]

2) When I  un-check the RTL checkbox, the mission is sent successfully to vehicle but when "Start Mission" command is given it shows error:Unable to start mission:Vehicle not ready.

[image]

These operations work fine  while testing on jMAVSim SITL which is running PX4 version 1.6.5dev and QGC compiled from master branch.

I'm using Intel aero board with intel-aero image 1.4 flashed on it.
BIOS_VERSION = Aero-01.00.12_Prod
OS_VERSION = Poky Aero (Intel Aero Linux Distro) 1.5.0-dev (pyro)"
AIRMAP_VERSION = 1.8
FPGA_VERSION = 0xc0

The firmware version as showed by QGC is 1.6.0dev.
[image]

sugnanprabhu, 2017-07-13:
@SaliniVenate 
The latest aero release is using v1.6.3 firmware.
Can you please try updating the px4 firmware on the aero with the following command and let us know if you still face the error.
# aerofc_update.sh /etc/px4-fw/nuttx-aerofc-v1-default.px4

anselmolsm, 2017-07-13:
> OS_VERSION = Poky Aero (Intel Aero Linux Distro) 1.5.0-dev (pyro)"
It is not 1.4.

> The firmware version as showed by QGC is 1.6.0dev.
Please follow what @sugnanprabhu suggested. That will flash 1.6.3 into Aero FC. Then repeat the test and post results here. Thanks.

lucasdemarchi, 2017-07-13:
There are lots of changes from v1.6.3 to v1.6.5. We are likely going to update once it's flight tested.  Can you test this firmware on aero? http://px4-travis.s3.amazonaws.com/Firmware/stable/aerofc-v1_default.px4

SaliniVenate, 2017-07-14:
I tried the updating px4 the way @sugnanprabhu  suggested and the issue 1 is solved. I'm able to send the mission successfully to vehicle. @lucasdemarchi , I tested the firmware you mentioned on intel aero and with that also I did not find any issue with sending a mission to the vehicle. In both cases the firmware version as displayed in QGC is 1.6.2dev.

The second issue  I believe is because the vehicle is not able to connect GPS as I'm working inside the lab.

lucasdemarchi, 2017-07-14:
Thanks for testing. Let's close this then.

closed 2017-07-14 by lucasdemarchi