intel-aero/meta-intel-aero #214: dockerfiles: Provide setups for some scenarios

anselmolsm, 2017-07-13:
Dockerfiles can be used to create docker images with dependencies of certain project pre-installed, for example. Basic ROS configuration is an example of useful dockerfile to have.

We'll setup a separate repo for those files (and put the link here when it's ready).

shakthi-prashanth-m, 2017-08-28:
@anselmolsm: While `docker pull ros` works well, am interested to know how does having `Dockerfile` for ROS is different ? Projects that use ROS may provide `Dockerfile` to build it using `docker`.

anselmolsm, 2017-08-28:
@shakthi-prashanth-m a Dockerfile can contain list of steps needed to install and configure certain application dependencies, for example. It's simpler to tell people to run docker build instead of a list of steps.