intel-aero/meta-intel-aero #218: Connect to ROS in docker

Nickhodem, 2017-07-19:
Is it possible to connect to ROS master which is set up in docker (https://github.com/intel-aero/meta-intel-aero/wiki/New-tools-v1.4) via local network?

anselmolsm, 2017-07-19:
With `docker run`, you can use either the `-p PORT` flag, which allows you to specify a port/list of ports to be exposed, or `-P` which exposes all ports.

On Aero, with ifconfig you can set an address (compatible with your local network) to docker0 interface. Then (assuming connected via wifi interface - wlp1s0):

```
iptables --append FORWARD --in-interface wlp1s0 -j ACCEPT
iptables --table nat --append POSTROUTING --out-interface wlp1s0 -j MASQUERADE
```

That should allow you to access the docker container from another machine in the local network.
We'll run a few tests with that and add those instructions to the docs. If you find any issue with them, please let us know.

Nickhodem, 2017-07-20:
Thanks for help! It works! I only had to export ROS_HOSTNAME = inet addr of wlp1s0.

anselmolsm, 2017-07-20:
Cool, @Nickhodem 

**For the record**: In fact, according to Docker's doc [Bind container ports to the host](https://docs.docker.com/engine/userguide/networking/default_network/binding/ ), the iptables commands I posted above are not needed. Thanks @lucasdemarchi for pointing that out.

Info about port binding also added to the [wiki](https://github.com/intel-aero/meta-intel-aero/wiki/08-Aero-Network-and-System-Administration#container-network-connectivity).

closed 2017-07-20 by anselmolsm