intel-aero/meta-intel-aero #219: rpm fails while checking certain types of signatures

anselmolsm, 2017-07-19:
This affects v1.4.

It might be related to build configuration parameters or dependencies. The same version of rpm works fine in other environments.

rpm operations that involves signature verification fail when the rpm file is signed using a RSA2048 key (however I got success while testing a RSA2048 key I generated myself. I'm still checking any other difference).

For example, capsule-01.00.12-r0.core2_64.rpm (which contains the BIOS update):

```
rpm -ivh capsule-01.00.12-r0.core2_64.rpm
error: capsule-01.00.12-r0.core2_64.rpm: signature region 62: tag number mismatch 7 ril 6 dl 792 rdl 168
```

Running a query for package contents results in an additional error:

```
rpm -qlp capsule-01.00.12-r0.core2_64.rpm
error: capsule-01.00.12-r0.core2_64.rpm: signature region 62: tag number mismatch 7 ril 6 dl 792 rdl 168

error: capsule-01.00.12-r0.core2_64.rpm: not an rpm package (or package manifest) 
```

No problem was verified while testing the same .rpm file with rpm 4.13.0 and building [the version packaged by Yocto](http://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/recipes-devtools/rpm/rpm_git.bb - sha1 a8e51b3bb05c6ac) on Fedora 26 and ArchLinux.

sugnanprabhu, 2017-07-20:
I'm trying to build the rpm from source on Ubuntu.
Following are the dependency list for rpm. Can you please share the dependency package version on the fedora where rpm -qlp command is working?

db >= 5.3.28
db >= 5.3.28
file >= 5.30
file >= 5.30
libarchive >= 3.2.2
libarchive >= 3.2.2
libbz2-1 >= 1.0.6
libbz2-1 >= 1.0.6
libc6 >= 2.25
libc6 >= 2.25
libdbus-1-3 >= 1.10.14
libdbus-1-3 >= 1.10.14
libdw1 >= 0.168
libdw1 >= 0.168
libelf1 >= 0.168
libelf1 >= 0.168
liblzma5 >= 5.2.3
liblzma5 >= 5.2.3
libpopt0 >= 1.16
libpopt0 >= 1.16
libz1 >= 1.2.11
libz1 >= 1.2.11
nss >= 3.28.1
nss >= 3.28.1
rpmlib(CompressedFileNames) <= 3.0.4-1
rpmlib(PayloadFilesHavePrefix) <= 4.0-1

lucasdemarchi, 2017-07-20:
Just submitted the bug upstream: https://github.com/rpm-software-management/rpm/issues/270

I think for now we should revert to rpm version 4.13.0.1 since it's the one used by Fedora (and thus more broadly tested) and also the last stable version.

anselmolsm, 2017-07-20:
> No problem was verified (...) building [the version packaged by Yocto](http://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/recipes-devtools/rpm/rpm_git.bb - sha1 a8e51b3bb05c6ac) on Fedora 26 and ArchLinux.

Something went wrong when we intended to test sha1 a8e51b3bb05c6ac . Test repeated today was able to reproduce the issue on Fedora.

Next step: Package rpm 4.13.0.1 for Aero (submit that to Yocto as well)

sugnanprabhu, 2017-07-21:
@lucasdemarchi @anselmolsm 
I made the changes to include rpm 4.13.0.1, following are the changes.
https://gist.github.com/sugnanprabhu/3945e137f82c7bf14a2f1b51dc3c9603

Also I have not included the the patches that are part of the fedora rpm package.

But facing the following error for all the packages. 

rpmlib(ShortCircuited) <= 4.9.0-1 is needed by kbd-keymaps-2.0.4-r0.corei7_64
rpmlib(ShortCircuited) <= 4.9.0-1 is needed by alsa-states-0.2.0-r5.corei7_64
rpmlib(ShortCircuited) <= 4.9.0-1 is needed by linux-libc-headers-dev-4.10-r0.corei7_64

anselmolsm, 2017-07-21:
@sugnanprabhu thanks for checking that. 

The current rpm recipe applies a patch that deals with that 'short circuit' thing. Besides that, there are other changes needed to downgrade to 4.13.0.1-release.

I don't think we are able to forecast all possible issues of such change. That said, is looks safer to follow the suggestion of ["back out that specific "exact_match" change"](https://github.com/rpm-software-management/rpm/issues/270#issuecomment-316893255), which means, apply a patch on top of the existing rpm recipe.

sugnanprabhu, 2017-07-24:
@anselmolsm @lucasdemarchi tried the changes suggested by rpm community.

https://gist.github.com/sugnanprabhu/45ba2aa653a80b415bcaf2a9c6572b86

Observing following error on installing the BIOS.

[image]

anselmolsm, 2017-07-25:
@sugnanprabhu The error above is not related to the rpm issue itself.

sugnanprabhu, 2017-07-25:
The issue caused earlier was due to installing the same version of BIOS.
@anselmolsm I have submitted PR #229

anselmolsm, 2017-07-26:
I've tested this and things seem to work with signed rpms. @lucasdemarchi any comments?

closed 2017-07-28 by anselmolsm