intel-aero/meta-intel-aero #220: AERO in Qgroundcontrol

gjkunde, 2017-07-19:
Dear All,

in summary QGC displays 

System ID    1
Airframe type <empty>
Vehicle type <empty>
Firmware version 1.6.2dev

what would need to be done in order to populate these fields ? Is that
at problem on the AERO side or the QGP side ?

Same question for updating firmware, it would be nice if there would be
a button like there is for PIXHAWK to update the firmware, rather than
copying the firmware on the machine and running aerofc-update.sh
Is that at problem on the AERO side or the QGP side ? What would need
to be done ?

Respectfully Gerd

anselmolsm, 2017-07-19:
I'm not sure about the first question, @lucasdemarchi may know the answer.

The second question, there is #202 somehow related to that.

gjkunde, 2017-07-19:
Indeed, but only somewhat. I read in #202, that
the idea is still get the file from /etc ... on the drone.
However for the pixhawk it automatically goes out to the
repository, finds standard, or beta or daily build.
(One can define a custom location too, I for example
compile it on my local machine), downloads  and then 
loads the firmware ... transparent and convenient :-)

anselmolsm, 2017-07-19:
@gjkunde your suggestion would be the "item c" in https://github.com/intel-aero/meta-intel-aero/issues/202#issuecomment-316471386 ;)

lucasdemarchi, 2017-07-19:
@gjkunde our first attempt to get the same behavior as in pixhawk in QGC was rejected by upstream developers: https://github.com/mavlink/qgroundcontrol/issues/4698

Our current strategy is to implement something similar to what has been discussed there:  an http api that QGC can use to update the flight controller.

Note that if you are building the code yourself, it's already possible to upload directly from your computer:

For PX4:
```
make aerofc-v1 upload
```

For ArduPilot:
```
./waf configure --board aerofc-v1
./waf copter --upload
```