intel-aero/meta-intel-aero #221: Pressure Sensor and Mag Sensor

gjkunde, 2017-07-19:
I am trying 
i2cdetect -y -r 2
and get a response 12 at address 0x12. According to:

Intel Aero Compute Board includes BMM 150 sensor. This sensor is connected to Atom SoC via I2C interface on bus 2 at slave address 0x12.

that should be the mag sensor - alas in QGC it says "mag0 not found".

I read in http://www.mouser.com/ds/2/783/BST-BMM150-DS001-01-786480.pdf on
page 16 that there is a self test. Is there a script available to do this ?

There is no response from address 0x76 (the only entry between 0x03 and 0xf was the magnet)
and the there is no pressure reading in QGC, alas I read: 

Intel Aero Compute Board includes MS5611 pressure sensor. This sensor is connected to Atom SoC via I2C interface on bus 2 at slave address 0x76.  

http://www.te.com/commerce/DocumentDelivery/DDEController?Action=showdoc&DocId=Data+Sheet%7FMS5611-01BA03%7FB1%7Fpdf%7FEnglish%7FENG_DS_MS5611-01BA03_B1.pdf%7FCAT-BLPS0036

lucasdemarchi, 2017-07-19:
Humn... are you sure you want to get the values from the sensors in the Compute Board side and not on the Flight Controller side?

lfelipe, 2017-07-19:
@gjkunde the Aero RTF has two separate boards inside, one Compute Board and one Flight Controller, and both of them have sensors available. 

This document should give you more info:
https://www.intel.com/content/www/us/en/support/drones/development-drones/000023602.html

If you are looking for source code examples of using those sensors, the following answer on the community forum should be helpful:
https://communities.intel.com/message/461760#461760

gjkunde, 2017-07-26:
Thank you for the comments, alas they do not address my question.
Are you implying that the Sensor section online is wrong ? It says:

_Command to access the Magnetometer Sensor

Intel Aero Compute Board includes BMM 150 sensor. This sensor is connected to Atom SoC via I2C interface on bus 2 at slave address 0x12. In order to communicate with this sensor, please use below linux i2c commands:

    root@intel-aero~# i2cdetect -y -r 2
    root@intel-aero~# i2cset -y 2 0x12 DATA-ADDRESS [VALUE] ... [MODE]
    root@intel-aero~# i2cget -y 2 0x12 [DATA-ADDRESS [MODE]]
Command to access the Pressure Sensor

Intel Aero Compute Board includes MS5611 pressure sensor. This sensor is connected to Atom SoC via I2C interface on bus 2 at slave address 0x76. In order to communicate with this sensor, please use below linux i2c commands:

    root@intel-aero~# i2cset -y 2 0x76 DATA-ADDRESS [VALUE] ... [MODE]
    root@intel-aero~# i2cget -y 2 0x76 [DATA-ADDRESS [MODE]]_

I get response from neither sensor trying the commands above, can you please
verify those ? Is the bus number right ?

After trying the commands on both RTF drones we have both of them are broken in
that QGC says that mag0 is no longer found. The pressure sensor was never visible.

Please help by checking that your boards behave according to your writeup and
please advise on what to do with the two AERO drones I have - neither can fly because
mag0 is in the preflight check.

lucasdemarchi, 2017-07-26:
@gjkunde as Luis said there are 2 boards:  the compute board and the the flight controller. The communication with QGC is from the flight controller, **not** the compute board.

The wiki is wrong wrt accessing the baro, we need to fix it. Thanks for reporting. MS5611 doesn't respond to I2C query, even if it's there.

Again, with RTF you should be looking for the sensors from the flight controller.

gjkunde, 2017-07-26:
Thank you ! Then let me rephrase again :-) 
Running the latest (in fact any) PX4 flight stack I neither
see the board magnet (it is on bus 2 with address 0x12 when
I do i2cdetect) nor the barometer. How would I start debugging
the problem ? I thought talking to them with i2cset and i2cget would
establish they are working ?

lucasdemarchi, 2017-07-26:
@gjkunde how are you running the PX4 flight stack?    It only runs on the flight controller and i2cset/i2cget is on the compute board.

What's the error message QGC is giving you?  Can you add a screenshot here on the summary tab?

gjkunde, 2017-07-26:
Thank you Lucas,

I compiled the stack after a git pull yesterday, copied it to the RTF and loaded it

gjkunde$ make aerofc-v1_default
[243/243] Generating aerofc-v1_default.px4, firmware_nuttx.bin

I run BIOS AERO-01.00.12_PROD
OS version 1.4.0 (pyro)
AIRMAP 1.8
FPGA     0xC0

attached is the QGC screen shot. I cannot fly anymore because mag0 is
a prerequisit. I tried calibrating the compass - after all orientations pass
successfully the calibration fails for mag0 is not there

[image]

I am planning to hook up lidar and sonar on the I2C bus - if you
say I cannot use I2C commands to talk to them how will I establish that
I correctly wired them into the bus ?

zehortigoza, 2017-07-31:
@gjkunde You can enable CONFIG_SYSTEM_I2CTOOL and use i2ctools in the Nuttx terminal to do similar things that i2cset/i2cget/i2cdetect... tools do.
To get into Nuttx terminal just connect to Aero board by Wifi or USB-over-ethernet, close any application that is using the udp port 14550 on the host machine and on the PX4-firmware directory run: Tools/mavlink_shell.py 0.0.0.0:14550

About the "no mag sensor 0" error, please try to connect to Nuttx terminal and run "ls /dev/", you should have a ist8310_ext in the list otherwise it can be a connection problem.

gjkunde, 2017-08-01:
@zehortigoza Thank you, I edited nuttx-configs/aerofc-v1/nsh/defconfig, did a make and
scp to the machine and loaded the FC - alas it still has no i2c ? What did I miss ?

$ grep CONFIG_SYSTEM_I2CTOOL  nuttx-configs/aerofc-v1/nsh/defconfig
CONFIG_SYSTEM_I2CTOOL=y
$ make aerofc-v1_default
$ scp ./src/firmware/nuttx/aerofc-v1_default.px4 root@192.168.8.1:

I listed the devices as you suggested, I understood that mag0 should be
http://www.mouser.com/ds/2/783/BST-BMM150-DS001-01-786480.pdf
here is the output:

gjkunde, 2017-08-01:
ls /dev/
/dev:
 accel0
 adc0
 baro0
 console
 gyro0
 ist8310_ext
 led0
 mag0
 mpu9250_accel
 mpu9250_gyro
 ms5611_int
 null
 pipe0
 pipe1
 pwm_output0
 px4fmu
 ram0
 tap_esc
 ttyS0
 ttyS1
 ttyS2
 ttyS3
 ttyS4
 ttyS5

When I open QGC version 3.2 it however says **mag0** not found, on both
RTF drones we have, both with the latest OS, Bios, Firmware and aerofc as compiled today.\
When we flew them out of the box, mag0 was there. Is this an upgrade issue ?

Thanks Gerd

zehortigoza, 2017-08-01:
@gjkunde It was a clean build? I'm not sure if it builds again the Nuttx with changes in defconfig

There is 3 mags on the Aero RTF kit.

Intel Atom -> I2C -> BMM150
AeroFC -> I2C1 - > mag internal(I don't remember the model right now but this is not used by PX4 or Ardupilot)
AeroFC -> I2C2 -> ist8310(this is the external mag inside of GPS box)

So looks like the ist8310_ext is being detected by PX4... try do airframe reset in QGC and then calibrate all the sensors again.

gjkunde, 2017-08-01:
Actually the correct message  is the 'primary compass not found', the system thinks
that it is mag0. I can go through the calibration procedure and only at the
end it says 'Emergency: mag0 fail, TOUT' ....

gjkunde, 2017-08-01:
I had said 'git clean -f', is there anything beyond that I need to do before I say make aero ?
I think 'make archives' is no longer required ?

lucasdemarchi, 2017-08-01:
> Actually the correct message is the 'primary compass not found', the system thinks
> that it is mag0. I can go through the calibration procedure and only at the
> end it says 'Emergency: mag0 fail, TOUT' ....

after this message, can you reboot and see if everything works?

zehortigoza, 2017-08-01:
I use "make clean"

gjkunde, 2017-08-01:
Thank you @zehortigoza, the airframe reset did the trick. It still does a TOUT
after booting but then finds the magnet.

I am still interested in the i2c though ... next is I want to connect the lidar lite.

zehortigoza, 2017-08-01:
@gjkunde I also got this one TOUT right after the boot then it stops but it can fly normally.

About the lidar you can follow this right now: https://dev.px4.io/en/flight_controller/intel_aero.html#connecting-a-lidar-lite-range-finder

We have some change inline to enable one more I2C in AeroFC and export it on the unused telemetry pins. https://github.com/PX4/Firmware/pull/7070
If you want to test that, just remember to update the FPGA firmware.

gjkunde, 2017-08-01:
@zehortigoza Even after 'make clean' and seeing that it built nuttx I do not see the i2c
command in the nuttshell :-( Do you have any more suggestions - what am I missing ?
Is there an easy way to confirm that I am running the right shell ? When I say 'uname -a'
I get 0 0.0 arm aerofc ....

I was actually planning to have a down looking lidar-lite from garmin and 
left right forward backward - a total of four - maxsonar mb1242 
I built an adaptor for the i2c splitter from pixhawk and want to use the
same bus, is your answer implying that you will have a second i2c bus on
the telemetry connector ?

zehortigoza, 2017-08-01:
@gjkunde you can copy the code of the tool from Nuttx to PX4 and build it... just take a look at some of PX4 drivers.

Right now only have one external I2C that is shared between the external compass and the battery ADC.
As you are going to add 4 sonars it think that is better for you to already move to this new solution with one more I2C bus. On this solution the external compass will have his own I2C bus and other bus will be shared between battery ADC and anything connected to the unused telemetry pins.

gjkunde, 2017-08-02:
@zehortigoza Oi José, using the extra port sounds like a wonderful idea,
what is the new pinout and what is the firmware version ? I assume it is not 0xC0 ?

I am not sure I understand 'you can copy the code of the tool from Nuttx to PX4 and build it... just take a look at some of PX4 drivers.' I think I need some more pedestrian instructions.

I have a HERE+ question, would be use a new issue for that ?

Thanks so much for your support.

zehortigoza, 2017-08-02:
@gjkunde on the PR link, there a link to the PR with the FPGA firmware on that PR there is the pinout and all other information that you need. Btw it was merged in PX4 today.

Each driver is actually a application that run in his own process in Nuttx, so you can create a new application with the code of the I2C tool, then you can run it from the Nuttx terminal.

Unless it is related with this issue please open another one.

closed 2017-08-02 by lucasdemarchi