intel-aero/meta-intel-aero #239: latest build failing on missing file/directory systemd-boot/recipeinfo

blankered, 2017-07-27:
Hi,

I am trying to compile the image from scratch. It compiles 6983 packages but fails on the following package.

```
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing python function do_image_complete
NOTE: Executing write_deploy_manifest ...
DEBUG: Executing python function write_deploy_manifest
ERROR: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:write_deploy_manifest(d)
     0003:
File: '/home/cbdlab/intel_aero/poky/meta/classes/license.bbclass', lineno: 33, function: write_deploy_manifest
     0029:        'w+').write(output)
     0030:}
     0031:
     0032:python write_deploy_manifest() {
 *** 0033:    license_deployed_manifest(d)
     0034:}
     0035:
     0036:python license_creat/meta-intel-aero/recipes-core/images/intel-aero-image.bb:do_image_completee_manifest() {
     0037:    import oe.packagedata
File: '/home/cbdlab/intel_aero/poky/meta/classes/license.bbclass', lineno: 192, function: license_deployed_manifest
     0188:        man_dic[dep]["IMAGE_MANIFEST"] = True
     0189:        man_dic[dep]["PN"] = dep
     0190:        man_dic[dep]["FILES"] = \
     0191:            " ".join(get_deployed_files(dep_dic[dep]))
 *** 0192:        with open(os.path.join(lic_dir, dep, "recipeinfo"), "r") as f:
     0193:            for line in f.readlines():
     0194:                key,val = line.split(": ", 1)
     0195:                man_dic[dep][key] = val[:-1]
     0196:
Exception: FileNotFoundError: [Errno 2] No such file or directory: '/home/user/intel_aero/poky/build/tmp/deploy/licenses/systemd-boot/recipeinfo'

DEBUG: Python function write_deploy_manifest finished
DEBUG: Python function do_image_complete finished
ERROR: Function failed: write_deploy_manifest
```

lucasdemarchi, 2017-07-27:
What command are you using to build?

blankered, 2017-07-27:
bitbake intel-aero-image

anselmolsm, 2017-07-27:
@blankered Have you copied the environment from one user to another?

I see reference to /home/**cbdlab**/intel_aero/poky/meta/classes/license.bbclass
Then the error is due to missing file at /home/**user**/intel_aero/poky/build/tmp/deploy/licenses/systemd-boot/recipeinfo

blankered, 2017-07-28:
Yes, I changed the username in the posted error log. I thought I had changed all references when I posted it. Here is the vanilla version:

```
DEBUG: Executing python function write_deploy_manifest
ERROR: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:write_deploy_manifest(d)
     0003:
File: '/home/cbdlab/intel_aero/poky/meta/classes/license.bbclass', lineno: 33, function: write_deploy_manifest
     0029:        'w+').write(output)
     0030:}
     0031:
     0032:python write_deploy_manifest() {
 *** 0033:    license_deployed_manifest(d)
     0034:}
     0035:
     0036:python license_create_manifest() {
     0037:    import oe.packagedata
File: '/home/cbdlab/intel_aero/poky/meta/classes/license.bbclass', lineno: 192, function: license_deployed_manifest
     0188:        man_dic[dep]["IMAGE_MANIFEST"] = True
     0189:        man_dic[dep]["PN"] = dep
     0190:        man_dic[dep]["FILES"] = \
     0191:            " ".join(get_deployed_files(dep_dic[dep]))
 *** 0192:        with open(os.path.join(lic_dir, dep, "recipeinfo"), "r") as f:
     0193:            for line in f.readlines():
     0194:                key,val = line.split(": ", 1)
     0195:                man_dic[dep][key] = val[:-1]
     0196:
Exception: FileNotFoundError: [Errno 2] No such file or directory: '/home/cbdlab/intel_aero/poky/build/tmp/deploy/licenses/systemd-boot/recipeinfo'

DEBUG: Python function write_deploy_manifest finished
DEBUG: Python function do_image_complete finished
ERROR: Function failed: write_deploy_manifest
```

anselmolsm, 2017-07-28:
I haven't seen this error in my builds, nor in the CI (incremental and clean builds).
However @avinash-palleti faced the same problem. He managed to "fix" that running: `bitbake -c clean intel-aero-image`.

Experience says that sometimes the content of poky/build/tmp gets messy and a rm is handy. Calm down, as long as you keep poky/build/sstate-cache, a new build won't take soooo long, as bitbake will regenerate rpms using the cache (it won't compile again), then sysrootfs, then the image.