intel-aero/meta-intel-aero #246: Lidar Lite V3 on new telemetry port I2C does not respond

gjkunde, 2017-08-03:
I verified the functionality of my Lidar Lite V3 by connecting it to my I2C2 port
on Pixhawk 2.1 and enabled the QGQ parameter SENS_EN_LL40LS and can
successfully see the distance readings by looking at the MAVLink inspector and
examine current_distance under the Distance Sensor tab.

However, if I plug said sensor into the telemetry port of the INTEL AERO RTF 
running the 7-26 firmware (0xC1) and an aerofc_v1 built today, then QGC does not have a 
mavlink entry for distance. Logging into nsh I do not see a driver with ls /dev.

It is correctly wired, I checked with a scope that I see the clock on line 4 - but never see any 
data line level transitions - even if I trigger on SDA on line 5. Was it verified that SDA on the
new I2C works or maybe is the driver for LL40LS only looking on the original I2C bus ?

zehortigoza, 2017-08-04:
If did not update the PX4 with the changes to enable this additional I2C it will not work.

gjkunde, 2017-08-04:
As in the other issues you just close, I did at git pull for
the latest firmware, made aerofc and uploaded it. How
would I check that I have the latest PX4 ?

anselmolsm, 2017-08-04:
> How would I check that I have the latest PX4 ?

Please post the output of `git describe` executed in your clone of PX4/Firmware.

gjkunde, 2017-08-04:
v1.6.5-350-ge7b7b27

LorenzMeier, 2017-08-04:
@zehortigoza Do you have a pull request?

gjkunde, 2017-08-04:
@zehortigoza @LorenzMeier 
This is following up as well on the 'closed issue on NO GPS and ADC'
I cloned #249 - @zehortigoza private clone of the PX4
git describe gives
v1.3.0rc1-3922-ge7d801c
I uploaded it onto the RTF alas I cannot talk to it anymore,
neither via ground control nor with Tools/mavlink_shell.py 0.0.0.0:14550
Does your version have a different baud rate ?

zehortigoza, 2017-08-04:
@LorenzMeier 
> @zehortigoza Do you have a pull request?
This was already merged into master, I have done a backport over the last stable release here: https://github.com/zehortigoza/PX4Firmware/tree/tags/v1
.6.5%2Badditional_i2c_backported

@gjkunde I answered your mail.

gjkunde, 2017-08-04:
v1.6.5-416-ga2ef611 is what I got after trying a fresh clone

gjkunde, 2017-08-04:
@zehortigoza I tried the backported version in the link last night , it does not allow a mavlink connection, did you maybe change the baudrate ? 

But YEAH, the merged master works, I have a HERE+ plugged in with its magnet connected
to the I2C port and I have the Lidar light in the telemetry port and all their reading show
up in the mavlink inspector !

Obrigado :-)

zehortigoza, 2017-08-04:
@gjkunde not baudrates changes... so all is working for you now?! glad to hear that
could you close all the issues that you opened that have being solved?

gjkunde, 2017-08-04:
@zehortigoza it appears that the version at  #249  has mavlink that does not work.

The master works. And indeed all issues seem to be resolved. I sent you email with
offline questions for some guidance before I open more issues ;-)

closed 2017-08-04 by lucasdemarchi

jdiament95, 2018-02-26:
I'm running px4 v1.7.3 on the intel aero RTF. My lidar lite v3 is attached to the telemetry port and I'm having the same issue – not seeing the distance sensor in the mavlink inspector. Any thoughts about how to troubleshoot this?

gjkunde, 2018-02-26:
Dear José,

let me find my logbook - I am working on a different project at the moment.

Gerd