intel-aero/meta-intel-aero #247: FPGA Version 0xC1 has no adc0 in /dev/

gjkunde, 2017-08-03:
Dear All,

when I compared the nuttshell command 'ls /dev' 
for 0xC0 and 0xC1 I got identical list but for the 'adc0',
it is not present in 0xC1. Is that the battery adc on the
I2C bus ?

gjkunde, 2017-08-03:
In addition to the missing adc0 there is no GPS information
visible in QGC, neither in the icon nor in the instrument panel.

Reverting back to 0xC0 restores both GPS and adc0.

zehortigoza, 2017-08-04:
https://github.com/intel-aero/meta-intel-aero/issues/221#issuecomment-319520005

you need to update PX4 as well.

closed 2017-08-04 by zehortigoza

gjkunde, 2017-08-04:
I did a GIT pull of the PX4 firmware and made aerofc and uploaded,
what else am I missing ?

lucasdemarchi, 2017-08-04:
On nsh console type:
ver git

And report the version.

zehortigoza, 2017-08-04:
> I did a GIT pull of the PX4 firmware and made aerofc and uploaded,
what else am I missing ?

Always run a 'make clean' after you fetch new commits this will guarantee that any changes in defconfig will be applied.

gjkunde, 2017-08-04:
1.6.5 0 (17171712)

gjkunde, 2017-08-04:
v1.6.5-350-ge7b7b27

zehortigoza, 2017-08-04:
> FW git-hash: e7b7b27ef8b0dbf67b5ecda3b9d1c382a73193bc

The patches enabling it was merged after this patch.

zehortigoza, 2017-08-04:
If you want to try the firmware that is going to be part of our release try with this one: https://github.com/intel-aero/meta-intel-aero/pull/249

gjkunde, 2017-08-04:
Thank you, could you please add I2C to nuttx in your release ? I am going
to clone it and try.

gjkunde, 2017-08-04:
OK, I cloned https://github.com/intel-aero/meta-intel-aero/pull/249
git describe
v1.3.0rc1-3922-ge7d801c
I uploaded it onto the RTF alas I cannot talk to it anymore,
neither via ground control nor with Tools/mavlink_shell.py 0.0.0.0:14550
Does your version have a different baud rate ?