intel-aero/meta-intel-aero #248: HERE+ with Intel RTF

gjkunde, 2017-08-03:
Dear All,

I am interested in using the HERE+ GNNS with RTK for the
RTF drone to get cm precision instead of using the Yuneec.

I had mastered the pinout for the HERE (no plus) and can see the 
GPS info in the pull down menu and the info in the flight panel
in firmware 0XC0 and its magnet works as well (all it needed
was a JST-GH to two DF13 adaptation. 

Alas the mavlink inspector does NOT list the GPS as part of the data stream ?
What could be the problem here ?

What is needed to get the RTK going or does it automatically register
the different GNNS ?

guermonprez, 2017-08-04:
Hello

Not answering your question, but FYI : Pixhawk Pro 3 with ublox RTK solution was validated with the aero compute board.

LorenzMeier, 2017-08-04:
RTK is fully supported with PX4. If you connect the ground module and use QGroundControl it will all work. A pull request with a tutorial would be welcome.

https://dev.px4.io/en/airframes_multicopter/dji_flamewheel_450.html

gjkunde, 2017-08-04:
@LorenzMeier Thank you, could you - in a few words - elaborate who
the GPS selection is done and how I2C addresses are handled ? If
I switch from the yuneec to the proficnc, is there a command sequence
which queries which GPS is present ? Where would my students find that
code ? If I add lidar and sonar, how are the device addresses handled ?
I know the lidar default is 0x62, the max sonar is 0x70, for I have four
I called them 0x70 2 4 6. It there a detect on the I2C and then a pull for
possible modules ? Again, where would we find the code ?

PS. Did you see my PX4 question about the d800 x 8 ?

zehortigoza, 2017-08-04:
The GPS should work fine without changes, about the compass you will need to add compass inside of HERE+ to the init script of Aero https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rc.sensors#L319

lucasdemarchi, 2017-08-04:
@zehortigoza just like we did with HMC5883 we can do for other mags on the external bus.  I need to confirm what's the mag on HERE+ but I think it is the lis3mdl

lucasdemarchi, 2017-08-04:
It also comes with a ms5611 that we could use... unless we are going to saturate the bus with that.

gjkunde, 2017-08-04:
@lucasdemarchi  I tried to find the code for the pixhawk2.1 which comes with
the HERE and the HERE+, there is seems to use either the HMC5883 or  lis3mdl,
the aero has the HMC5883 start but with different options ? Maybe you can help here ?

lucasdemarchi, 2017-08-04:
I think the first production runs were using hmc5983 but now they are switching to lism3dl since the former is EOL. If yours is hmc it should just work

zehortigoza, 2017-08-04:
> @zehortigoza just like we did with HMC5883 we can do for other mags on the external bus. I need to confirm what's the mag on HERE+ but I think it is the lis3mdl

The problem in enabling several mags in PX4 upstream is that someone could just connect another mag and try to fly with a wrong mag rotation.

gjkunde, 2017-08-04:
My code lines in the new init.d are:
< 	#if hmc5883 -X start
< 	#then
< 	#fi
<
<         if hmc5883 -C -T -X start
< 	then
<         fi
<
< 	if lis3mdl -R 2 start

I see the following but is says mag0 self test fails:

nsh> hmc5883 info
WARN  [hmc5883] running on bus: 2 (/dev/hmc5883_ext)

hmc5883_read: 14318 events, 15355416us elapsed, 1072us avg, min 867us max 3988us 406.869us rms
hmc5883_com_err: 0 events
poll interval:  20 ticks
output  (0.21 -0.21 0.00)
offsets (0.00 0.00 0.00)
scaling (1.00 1.00 1.00) 1/range_scale 820.00 range_ga 1.90
temperature 0.00
report queue	2/112 (0/2 @ 1000b9a0)

gjkunde, 2017-08-04:
We checked the proficnc webpage it only says that the HERE has
the two we discussed above but that the HERE+ has 4 (alpha testers only)
without actually specifying which one is in the shipped version, I will contact them.

gjkunde, 2017-08-04:
Thank you for the reset airframe tip. There are no longer
self test errors. Only warning after calibration is that 
there is a large offset between magnet and GPS, I have
to check the orientation - thanks for all your help !

lucasdemarchi, 2017-08-04:
> The problem in enabling several mags in PX4 upstream is that someone could just connect another mag and try to fly with a wrong mag rotation.

For external mags the user can mount it with whatever orientation he wants, as long as he informs the orientation. Usually external mags have an arrow to allow easy placement.

gjkunde, 2017-08-05:
I know and that is why I am confused ... I use the same HERE+ on the pixhawk,
I thought if I orient it the same (it has a clear arrow) and initialize it as in pixhawk
then it would come out aligned :-)

gjkunde, 2017-08-09:
sales@hex.aero says:

HERE+ has four magnetometers: HMC5983, LIS3MDL, AK09916 and MPU9250.

gjkunde, 2017-08-18:
Who will put the magnetometers into init.d for the official release ?

closed 2018-06-05 by zehortigoza

suraj-bijjahalli, 2019-06-24:
@gjkunde Hello I am attempting to install a similar setup on the RTF.
I am new to this and am having trouble with the choice of connectors to interface the rover unit with the rtf. Could you please tell me how exactly you connected the rover unit. Did you have to adapt some cables?
Any pointers would be very helpful. Thank you.