intel-aero/meta-intel-aero #250: GPS unit is turnable on mast (potentially causing issues for the flight controller)

LorenzMeier, 2017-08-04:
We've had a few units where the top GPS cover is loose. A direction change there is problematic with the in-built compass and could cause toilet-bowling or other symptoms of deteriorated flight performance.

lucasdemarchi, 2017-08-04:
Yep. This is a known issue. We should add that to the new docs and suggest how to fix it. Thanks.

suraj1988, 2018-08-17:
Hello. Im currently experiencing similar issues on my quad which i suspect may be due to this.  Any suggestions for working around this ?

lucasdemarchi, 2018-08-17:
Make sure the screw is tight (note there's a small hex screw in the mast). Better yet put a little bit so it doesn't turn.

lucasdemarchi, 2018-08-17:
Closing this as it is a hardware limitation that can't really be fixed, but should have be workarounded.

closed 2018-08-17 by lucasdemarchi