intel-aero/meta-intel-aero #256: How connect to intel aero board with using USB cable

Aurusd, 2017-08-07:
I try connect to intel aero board with using USB cable via putty program but itsnt successful. Thera are another way or what i missed?

lucasdemarchi, 2017-08-07:
What version of Aero are you using? What OS objetivos your computer?

Aurusd, 2017-08-07:
INTEL® AERO READY TO FLY DRONE with 4.4.3 yocto standart

Aurusd, 2017-08-07:
Operating System | Yocto Project 2.1 (Krogoth), Linux* 4.4.3-yocto-standard

Aurusd, 2017-08-07:
I use usb3.0 to usb or microUSB to USB cables, board detect like COM3 on PC, but i cant make connection via putty

avinash-palleti, 2017-08-07:
@Aurusd Are you getting new network interface in your host machine when you connect to Aero Board using USB cable?

If Yes, then it is ready to connect through SSH. If No, there should be some issue with either cable or usb drivers on host machine. Can you help us providing dmesg logs from your host machine after you connect USB Cable.

To get OS version you can run aero-get-version.py on Aero board which is available by default.

lucasdemarchi, 2017-08-07:
> Operating System | Yocto Project 2.1 (Krogoth), Linux* 4.4.3-yocto-standard

This is before Aero 1.2.  In that version it doesn't get an IP via dhcp so you will need to set it manually.  However, you should rather update Aero:
https://intel-aero.github.io/getting-started/update

I was also asking by your host OS (i.e. the one on your computer/laptop).  The usb-ethernet works on Linux and Mac. On Windows it may or may not work depending on drivers installed.

lucasdemarchi, 2017-08-07:
See #154

Aurusd, 2017-08-07:
When i used aero-get-version.py - result is nit successful, command not found

lucasdemarchi, 2017-08-07:
See the link to the update section above. If you don't have this command it means you have version before 1.2

Aurusd, 2017-08-07:
On my windows PC, when i connect aero board with using usb cable, windows detected com port COM3, but putty isnt connect via this port

Aurusd, 2017-08-07:
ok, thanks lucasdemarchi, i will try tomorrow to update

Aurusd, 2017-08-08:
i have been install 1.4 version and activate wifi on aero drone. Have connection via ssh but still dont understand how to connect to drone via usb cable?

Aurusd, 2017-08-08:
and another question, how can i install java on drone? use this link https://thelinuxfaq.com/473-yocto-how-to-install-java-jdk-in-intel-edision-board on edison, its work fine but on aero  it isnt work, have you any advices?

avinash-palleti, 2017-08-08:
@Aurusd Can you please provide the error you got while following the installation steps.

And regarding ssh via USB cable, on your windows machine did you install RNDIS Ethernet driver because that's what it is asking me to install if i connect to my windows pc.
If Yes, Can you provide ifconfig output on Aero board.

Aurusd, 2017-08-08:
[image]

Aurusd, 2017-08-08:
if i connect usb cable i have never see any message for install something
[http://clip2net.com/s/3MM2ieP](url)

Aurusd, 2017-08-08:
via ssh with usin usb cable connect yo 192.168.7.2 isnt successful

avinash-palleti, 2017-08-08:
@Aurusd As ifconfig on Aero showing that USB interface is up and ip-address is also assigned to it, so there is no issue from Aero side. Below is the screenshot what i see when i connect Aero to windows pc.

[image]

And also from screenshot i can see you are trying to connect through serial interface in putty. Can you give try with SSH and with 192.168.7.2 address and root as username.

lucasdemarchi, 2017-08-08:
@avinash-palleti what version of Windows are you using? Can you take a look on the doc section about usb on windows and correct any problem there?

avinash-palleti, 2017-08-08:
sure @lucasdemarchi

Aurusd, 2017-08-08:
Windows 10, and i also try 192.168.7.2 address and root as username instead of serial interface, but still no connection. When connect aero vai usb cable there isnt any other device. I think its maybe because win10 use some standard driver

Aurusd, 2017-08-08:
i have try driver toolkit program, and saw that there isnt driver for USB/LPT port (RNDIS/Ethernet gadget), but program didnt download it.  Maybe its problem on win 10

Aurusd, 2017-08-08:
its something like here https://www.mobileread.com/forums/showthread.php?p=3283986
On Windows 10, when jailbroken Kindle with enabled USBNet is plugged via USB, it will appear as a "Serial USB device (COM3)" in device manager.
If Windows 10 already recognized your Kindle device as an extra network card, you do not need this driver package :3
i have download driver files from http://www.driverscape.com/download/rndis-ethernet-gadget and do the same action but driver isnt update

avinash-palleti, 2017-08-09:
@Aurusd I am using windows 8.1, after installing RNDIS driver, in device manager you should see new device in network adapters section like below,

[image]

I took driver from the same link which you shared above, http://www.driverscape.com/download/rndis-ethernet-gadget.
Please follow driver update process as mentioned in below link,
https://www.lifewire.com/updating-drivers-in-windows-7-tutorial-2619224

After that i am able to do ssh using putty from my windows PC.

[image]
[image]

Aurusd, 2017-08-09:
try today on win7.. everything work fine like you described before, on win 10 no device.
[image]

lucasdemarchi, 2017-08-09:
Solution seems to be here:
https://answers.microsoft.com/en-us/windows/forum/windows_10-networking/windows-10-vs-remote-ndis-ethernet-usbgadget-not/cb30520a-753c-4219-b908-ad3d45590447

tldtr: a bug in windows 10 that doesn't declare that usb descriptor as compatible with rndis.

For the current version (1.5.1) I think we should document the solution 1 described there:

> Solution 1: If you are one of the people that found this because your device is showing up as a COM port instead of RNDIS, you may be able to get away using the RNDIS 5.1 driver. Find your device under Ports (COM & LTP) in Device Manager. Right-click it and select Update Driver Software..., then Browse my computer for driver software and then Let me pick from a list of device drivers on my computer and finally choose Remote NDIS Compatible Device. This should install the Microsoft RNDIS 5.1 driver (shows "Acer" as the manufacturer). If this works great. If it causes Network and Settings and other network related things to lock up, then you need the RNDIS 6.0 driver instead. Try one of the other solutions.

For next version we may investigate solution 2:

> Solution 2: If you have control over the RNDIS device and it runs Linux (i.e. BeagleBone), you can tweak the driver to get along with Windows better. For example, if you have a 3.16 kernel or newer, you can setup your gadget using configfs and include os descriptors. This is actually what I ended up doing for my case. You can find my fully documented script here. The key was specifying the compatible and subcompatible ids so that it matches "USB\MS_COMP_RNDIS&MS_SUBCOMP_5162001" in rndiscmp.inf. This causes the Microsoft RNDIS 6.0 driver to be installed for this device. With this driver, I have not seen the same lockup as I have with the 5.1 (Acer) driver. I also changed the vendor and product ids so that it wouldn't match the "Acer" driver.

Btw I think that recent versions of Debian that comes on BBB and BB Blue is doing exactly that. We may take a look on how they are doing that. Just need to take care that the same USB port can be a host port.

Thanks all for the testing and feedback.

Aurusd, 2017-08-09:
Solution 1 didnt work..((

avinash-palleti, 2017-08-10:
@Aurusd Can you please confirm if any of the solution from above link worked for you.
> https://answers.microsoft.com/en-us/windows/forum/windows_10-networking/windows-10-vs-remote-ndis-ethernet-usbgadget-not/cb30520a-753c-4219-b908-ad3d45590447

Aurusd, 2017-08-10:
Like here it helps
Download & Unzip attachment kindle_rndis.inf_amd64-v1.0.0.1.zip
R-click "5-runasadmin_register-CA-cer.cmd" and "Run as administrator"*
In Device Manager, expand "Ports (COM & LPT)", R-click "Serial USB device (COM3)" > Update Driver Software...
Browse for my computer for driver software > Select extract folder
But! still no connection with aero drone via qGroundControl application(

sugnanprabhu, 2017-08-10:
@Aurusd 

Please try the following. I tried the following steps on Win 10 and see that the issue is resolved.

https://answers.microsoft.com/en-us/windows/forum/windows_10-networking/windows-10-vs-remote-ndis-ethernet-usbgadget-not/cb30520a-753c-4219-b908-ad3d45590447?auth=1

> `Thanks a lot Andrea !
Worked for me with the Raspberry Pi Zero.
Reddit redirects to a french tutorial that gives driver files, here are the direct links for future readers:
Tutorial: http://domotique.caron.ws/cartes-microcontroleurs/raspberrypi/pi-zero-otg-ethernet/
Driver files: http://domotique.caron.ws/wp-content/uploads/telechargement/RPI%20Driver%20OTG.zip
`
> `Just click "search my computer..." and give the path to the downloaded folder. The device will be recognized as a RNDIS Gadget.
`

sugnanprabhu, 2017-08-14:
@Aurusd Can you confirm if the issue is resolve with the above steps?

Aurusd, 2017-08-14:
help only this
Download & Unzip attachment kindle_rndis.inf_amd64-v1.0.0.1.zip
R-click "5-runasadmin_register-CA-cer.cmd" and "Run as administrator"*
In Device Manager, expand "Ports (COM & LPT)", R-click "Serial USB device (COM3)" > Update Driver Software...
Browse for my computer for driver software > Select extract folder

zehortigoza, 2017-09-29:
@Aurusd can the be closed? I think you already have a working setup.

lucasdemarchi, 2017-10-19:
I think this can now be closed

closed 2017-10-19 by lucasdemarchi