intel-aero/meta-intel-aero #259: Update aerofc-update.py script to give proper debug messages

avinash-palleti, 2017-08-10:
Script asks to unplug and replug the USB connector which is confusing,

https://communities.intel.com/message/491806#491806

diksha-gupta, 2017-08-22:
Upon tracing the root cause of the following warning message 
"Attempting reboot on /dev/ttyS1 with baudrate=115200...
If the board does not respond, unplug and re-plug the USB connector.",
it is found that it is an issue with boot loader SYNC - This is per px_uploader.py file which in turn being called by aerofc-update.sh script file.

Therefore it would be appropriate if we could print the message as "issue with the bootloader sync" from px_uploader.py script file.

Likewise it has to be followed for other info log messages too.
Please let us know your opinion on the same.

avinash-palleti, 2017-08-22:
px_uploader.py is from PX4, this script is originally developed to update the Firmware from host machine. So those log messages to reconnect/connect USB connector makes sense. 
If we are having the copy of that script in Aero image, we need to change all those messages accordingly.

> Therefore it would be appropriate if we could print the message as "issue with the bootloader sync" from px_uploader.py script file.

There might be issue with the connections to aeroFC. So it would be apt if we print check the connection to aeroFC and try again.

@lucasdemarchi Any thoughts?

lucasdemarchi, 2017-08-22:
Changing the messages will be bad for future updates. Talk to px4 upstream and check, explain the use-case and check if there's an acceptable change. The fix can be as simple as "I'm not trying to update via USB but rather via uart, so there's no reason to give that suggestion".

anitha-suresh, 2018-01-03:
@avinash-palleti , can you please bring up in the dev call and close this issue accordingly.

avinash-palleti, 2018-01-17:
https://github.com/PX4/Firmware/pull/8700
Changes are submitted to PX4, and PR is merged.
On our next Firmware update, we should able to see this fix.