intel-aero/meta-intel-aero #261: Can't connect to QGroundControl

Aurusd, 2017-08-11:
Update intel aero drone firmware to 1.4 version, like here https://github.com/intel-aero/meta-intel-aero/wiki/02-Initial-setup#flashing
But after this cant connect from qgroundcontrol to drone. Connect to drone AP wifi occurred   next problem
[image]
I am not find how can to see current mavlink baud rate of qgroundcontrol  app. Try to change mavlink baud rate on drone (in this file /etc/mavlink-router/main.conf), but error still present. 
Also try to use usb cable (RNDIS ethernet gadget) qgroundcontrol  show nothing. Test on win 7

zehortigoza, 2017-08-15:
I saw this issue that you opened in PX4 https://github.com/PX4/Firmware/issues/7765 maybe you loaded a PX4 firmware targeted to other board. If so please follow the recovery process: https://github.com/intel-aero/intel-aero-docs/pull/33 following method 2.
QGroundControl should connect to AeroFC without any changes on the mavlink-router configuration file.

If even after you load the PX4 firmware that comes with the image the communication with QGC did not work please run this and paste the output:
```
systemctl stop mavlink-router
mavlink-routerd -r
```

Aurusd, 2017-08-15:
When i bought drone i have version lower than 1.2, i then i have flashed the last version 1.4 https://downloadcenter.intel.com/download/26932/Intel-Aero-Platform-for-UAVs-Installation-Files
After i cant connect via QGC. 
ifconfig 
[image]
systemctl stop mavlink-router
mavlink-routerd -r
[log_mavlink_routerd.txt](https://github.com/intel-aero/meta-intel-aero/files/1224408/log_mavlink_routerd.txt)

zehortigoza, 2017-08-15:
By the logs there is some problem between Aerofc and mavlink-routerd. Did you changed the baudrate in the configuration file?
The latest version is 1.5.1v try update to that, including PX4, FPGA and BIOS of the same release.

Aurusd, 2017-08-15:
yes, i have changed baud rate, by default in configuration file was 468000, i changed to 57600. Ok i will try

Aurusd, 2017-08-15:
update for version 1.5.1. But even from phone app QGC cant connect, i saw drone icon on map, but on app have text - Waiting for Vehicle connection (connect to drone wifi AP, and put password 1234567890)

Aurusd, 2017-08-15:
i have said not true, on phone its work! but on win10 no

Aurusd, 2017-08-21:
for now it works!

closed 2017-08-21 by Aurusd