intel-aero/meta-intel-aero #262: Cant parse Mavlink message from flight controller

Aurusd, 2017-08-11:
Have program code which fine parse Mavlink message from PX4.  Try to use this code on aero drone, but cant parse mavlink message. Input buffer from ttys1 include  not much values. Use command 
/etc/init.d/mavlink-routerd.sh stop situation change, input buffer consists too much values, but also cant parse mavlink message. 
Can someone explain what is wrong? mavlink message on aero drone and PX4 must be same?

zehortigoza, 2017-08-15:
Take a look at https://github.com/01org/mavlink-router/tree/master/examples

Aurusd, 2017-08-15:
i think maybe difference between version of mavlink. PX4 has 2.0, intel aero works on 3.0

zehortigoza, 2017-08-15:
MAVLink 3.0 do not exist yet and IntelFC runs PX4 so it will be the same version.
Please try to run the examples first, they are tested, you can based your application on those examples.

Aurusd, 2017-08-16:
what is a main reason of mavlink-router? redirecting mavlink packets to QGC? what can i do if i want control drone both from my code and QGC, because if dont stop systemctl stop mavlink-router service, i cant see packets in my code, but if i stop, i cant connect to drone from QGC

lucasdemarchi, 2017-08-16:
mavlink-router routes packets between multiple endpoints besides having native support for saving flight stack logs.. If you change your code to use tcp or udp rather than uart, then you can let them to co-exist

closed 2017-08-16 by lucasdemarchi