intel-aero/meta-intel-aero #269: GPS isn't work

Aurusd, 2017-08-19:
No signal from GPS
[image]
GPS is turned up, and wire is connected, but there isnt any satellite.
How can resolve it or at least test what is caused this&

anselmolsm, 2017-08-22:
Just checking, have you tried that outdoors?

Aurusd, 2017-08-22:
yes i have tried outdoor, but there is no signal gps

guermonprez, 2017-08-24:
@anselmolsm what would be the best way to debug this problem? do we have more logs?
if QGC says the link with the drone is ok, I guess there is no way the GPS messages would be lost in the middle?
so next step would be to test far from buildings?
what do you think?

anselmolsm, 2017-08-24:
My question about testing outdoors was because I've seen GPS issues related to interference indoors in the past.
@lucasdemarchi @zehortigoza any suggestion regarding logs?

zehortigoza, 2017-08-24:
What is the PX4 version and the FPGA version?

To get the PX4 version:
Go to the icon with the paper and a magnifier(the 5th from the left) then go to Mavlink Console and type: ver all
please paste the output here

To get the FPGA version:
Go to Aero board terminal(by SSH or with monitor and keyboard) and type: aero-get-version.py

Why I'm asking this is because newer PX4 firmwares(including the one in the 1.5v image) needs the new FPGA firwmare. If your FPGA firmware is 0xC0 you need to update it, and if not update the problem will be exactly the one that you are having.

VineeshKK, 2017-08-29:
You can  quickly check if GPS coordinates are received by FC.
1.	SSH to Aero shell.
2.	Copy the _mavlink_shell.py_ to _/home/root_
a.	You can find **mavlink_shell.py**  here at https://github.com/PX4/Firmware/tree/master/Tools

3.	Then login into FC shell( nsh>) using the following command
a.	**./mavlink_shell.py 0.0.0.0:14550**
4.	Run the FC shell command:  **listener vehicle_gps_position** 

 It will show lat: lon: and other GPS parameters like hdop, vdop..etc.

Aurusd, 2017-09-04:
ver all
[image]

VineeshKK, 2017-09-04:
@Aurusd  Are you able to get GPS Coordinates from NutShell?

zehortigoza, 2017-09-05:
@Aurusd What about the FPGA version? With this PX4 firmware version you need the FPGA firmware with version 0xC1.

Aurusd, 2017-09-08:
oh.. sorry your advise is right and now gps works! Thanks!

closed 2017-09-08 by Aurusd