intel-aero/meta-intel-aero #271: Suggestion: Adding the package of python-devel

leixuchn, 2017-08-21:
When I want to install the dronekit from https://github.com/dronekit/dronekit-python, I find the head file of python.h is lacked. The instruction from the net is to install the package of python-devel. However,  python-devel is not found by DNF. Would anyone give some instruction about that ?

Rjasuja, 2017-08-21:
Dronekit can be installed using : pip install dronekit.
If you still want to install from the source code, push the rpm package for python-devel to aero dnf server. 
I am not sure about the mode dependencies.

trgiman, 2018-06-11:
Hi all. Is this still in progress. For instance. I can not install pip install zbar. Because Python.h is missing.