intel-aero/meta-intel-aero #273: Suggestion: adding tmux

JonasVautherin, 2017-09-08:
Is it already possible to install tmux on the Aero? If not, would it be possible to add the package?

anselmolsm, 2017-09-08:
Hi @JonasVautherin , tmux package is not available yet, we'll keep this issue updated about that.

As an alternative in the meanwhile, for terminal multiplex (with less features than tmux)  there's `screen` which is already in the image.

JonasVautherin, 2017-09-08:
Yeah, I know about screen. I just find tmux more convenient.
This is just a suggestion =). Feel free to close whenever you want.

Thanks a lot!

adityavarmatcs, 2017-10-06:
Recipe for tmux package is already present in the layer meta-openembedded/meta-oe.
We can insert this package into the image by adding IMAGE_INSTALL_append = " tmux" to the local.conf file.

closed 2018-07-29 by JonasVautherin