intel-aero/meta-intel-aero #276: Connect 4g modem telit ln930

Aurusd, 2017-09-22:
Have any body some experience or tutorial how to configure and use telit ln930 or same m2 modem with intel aero drone?

guermonprez, 2017-09-22:
should be online today or monday, thanks

Aurusd, 2017-09-24:
On my computer board version i have plug modem without problem. But on yocto linux 1.5.1 i haven't seen any telit device (no lsusb, no lspci). What i have missed? maybe i need to build my own yocto image with ither drivers. Have anybody advices? maybe there isn't some drivers, cause this modem not also not detected

Aurusd, 2017-09-24:
on this guide https://www.intel.com/content/dam/support/us/en/documents/boardsandkits/aero/aero-platform-uavs-mechanical-assembly-guid…  i have connect telit modem to R RESERVED for future use maybe itsn't m.2, cause on 13 M.2 Interface for PCIe SSD there is pins mismatch beetwen telit modem and computer board

Aurusd, 2017-09-24:
maybe i need to activate R RESERVED or its not work actually?
[image]

Aurusd, 2017-09-24:
And there is a big question second m2 socket type B marked on https://www.intel.com/content/dam/support/us/en/documents/boardsandkits/aero/aero-platform-uavs-mechanical-assembly-guide.pdf​
like R Reserved and socket for sim card actually is active or empty?

broody, 2017-09-24:
Hi Ausurd, have you updated BIOS as well? v01.00.13 is needed.

avinash-palleti, 2017-09-25:
@Aurusd As @broody mentioned v01.00.13 bios is required to get Modem working. Below are the slots for both modem and SIM card,

[image]
You actually need a SIM card holder one like below to insert into respective slot,
[image]

Further if you face any issues in connecting to live network, you might need to update APN as per your operator. Please find the steps to update APN at https://github.com/intel-aero/meta-intel-aero/wiki/08-Aero-Network-and-System-Administration#networking-lte

Aurusd, 2017-09-25:
Ok.. thanks for advise i will try.

Aurusd, 2017-09-25:
Modem now detect correctly!!)) There is somethere document which describe bios version firmware (what support and what no) or i can only solve my problem by publish issue on github?

lucasdemarchi, 2017-09-25:
Only the latest bios (1.00.13) with latest bsp (1.5.1). FC firmware can be any one

closed 2017-09-25 by lucasdemarchi