intel-aero/meta-intel-aero #278: Intel Aero RealSense Functionality

jonly123, 2017-09-27:
Recently purchased the Aero RTF kit and am trying to learn and experiment with the RealSense and camera hardware that it came with.

I am able to stream from the camera in the realsense module following the "Getting Started" guide for the RTF, but I cannot figure out how to view video feed from the bottom and side mounted camera.

I have read through a handful of threads and found some sample code (links below) but I don't exactly know how to load/compile the sample apps onto the Aero and implement them.  Any guidance or reference materials would be very helpful.

https://github.com/zehortigoza/aero-optical-flow
https://github.com/intel-aero/sample-apps
https://communities.intel.com/thread/112209
https://github.com/intel-aero/meta-intel-aero/issues/223

zehortigoza, 2017-09-28:
If you just want to stream you can do that by rebuild the Intel Aero image, this patch https://github.com/intel-aero/meta-intel-aero/commit/152d250611a4ab3c82a861a22e2635eb7b6c385a is enabling the stream of the bottom camera. Look at the Camera Streaming Daemon for more information https://github.com/01org/camera-streaming-daemon/wiki/Quickstart-Guide

If you want to build the capture application from sample-apps here is the makefile so you just need to type "make" to build it.

The Intel Aero image may not have some libraries that you will need to build applications using the camera data in this case you can build those libraries in Yocto and copy the RPM to the Aero and installing then or you can use Docker.

jonly123, 2017-10-02:
Thanks for the links -

I am not very familiar with the Linux/Ubuntu so bear with me!

I have read through the Daemon and understand it at a basic level.  I will set up a Ubuntu VM on my laptop for testing.

**If you want to build the capture application from sample-apps here is the makefile(intel-aero/sample-apps#3) so you just need to type "make" to build it.**

If you could just elaborate slightly on this.  I am assuming I need to copy the folder/files(?) over to the Aero, navigate to the directory and then run "make" on the file?

zehortigoza, 2017-10-02:
> If you could just elaborate slightly on this. I am assuming I need to copy the folder/files(?) over to the Aero, navigate to the directory and then run "make" on the file

Exactly, you can also try to build in your host machine and copy the binary to Aero, for this it should work other application may not work because your host machine could have a different version of some libraries than Aero have.

jonly123, 2017-10-02:
How do I know what file to copy over for the sample apps?  There are a few files in the **capturev4l2** folder

zehortigoza, 2017-10-02:
If building in Aero you are going to need all files, if building in host machine you just need to copy the capture binary.

closed 2018-06-05 by zehortigoza