intel-aero/meta-intel-aero #279: BIOS Reset

dkircali, 2017-10-18:
I am using Ubuntu 16.04. USB 3 camera was being detected as USB2, so I was trying BIOS settings to solve it.
I have disabled  xHCI mode, it disabled only USB connection available. It is not detecting keyboard so I can't go back to BIOS to revert the settings or boot from USB to flash an official release to revert the changes.
Is there any way to reset the BIOS?

lucasdemarchi, 2017-10-18:
Can you still boot? If so you have some options:

1. Put a bios update on /boot (this is a .fv file that may be inside an rpm); this will trigger an update on next boot, but I'm not sure it will erase current options
2. Install efibootmgr and
  a.  use it to select the next boot entry
  b. use it to erase current variables
  c. use it to set OsIndications variable to 1 and thus trigger a "enter firmware interface" on next boot

dkircali, 2017-10-19:
@lucasdemarchi thank you for your response.
Board boots directly to Ubuntu then I can use remote desktop/ssh to control it from another PC.
1) I have tried putting .fv file in /boot folder. Nothing happens it goes directly (boots) to Ubuntu
2) I am not sure if it is a good idea to change boot order, if it won't boot to OS I won't have any options.
3) I can go to the first BIOS screen by typing this command from Ubuntu:
`systemctl reboot --firmware-setup`
However as no keyboard is detected I can not go any further

I really don't understand why there is no switch/jumper to reset bios.

lucasdemarchi, 2017-10-19:
The command with efibootmgr was changing only the **next** boot, not the boot order. I agree with you it's not a good idea to change the boot order.

I thought that after entering the efi interface you would be able to control it, but I was wrong.

Another possibility is to check if the setting is exported as an efivar and set it from Linux. However this may be risky of bricking your board. I can try something tomorrow on one of my boards.

I agree  with you there should be an easier option to reset the bios.

zehortigoza, 2017-10-26:
Can you use keyboard when grub(or your bootmanager) shows up? If so I think you could add a entry to boot from USB and install the Yocto image.

I guess Ubuntu created the /boot partition different than BIOS expected to run the auto-update.

zehortigoza, 2017-10-26:
What Lucas was asking you to try was this:
root@intel-aero:~# mount -t efivarfs none /sys/firmware/efi/efivars/
root@intel-aero:~# efivar -l | grep RestoreFactoryDefault
root@intel-aero:~# efivar -n <id above>-RestoreFactoryDefault -w 1
But after testing several combination of arguments I always get a invalid argument error.
Maybe this command works in Ubuntu, please give a try.

whitebrandy, 2017-11-16:
Sorry to hijack this thread but I have been trying this as well since I have ended up with a BIOS that I can't make changes to. 

*RestoreFactoryDefault has "Authenticated Write Access" in the attributes and according to the spec, the data buffer should begin with an authentication descriptor. This seems to be the reason as to why writing a 1 doesn't succeed.

I am pretty much stuck now,  wondering if there are any other workarounds to do a factory reset on the Aero board.

dkircali, 2017-11-21:
@zehortigoza USB is completely disabled. Only way I can interact with the the card is after booting to OS and using remote desktop connection. 

`root@intel-aero:~# efivar -n <id above>-RestoreFactoryDefault -w 1`

is giving invalid argument error as I believe write option does not exist.

I also asked help from intel community:
[https://communities.intel.com/message/509319#509319](url)
Flashing the original bios file has been suggested but I couldn't manage to do it.

@whitebrandy I am also stuck, and I have already lost my hope. Just keep in mind that Intel Aero Computing Board also can be used as a very nice, expensive **paperweight**.

closed 2018-06-05 by zehortigoza