intel-aero/meta-intel-aero #286: SD card functionality for Aero FC

lkm1321, 2017-10-24:
Hi, 
Is there a way to emulate a SD card plugged into the onboard FC? On PX4, this is required for the following functionalities which I really need: 
1) Polygonal geofencing
2) Extra rc.local for NuttX PX4
Further, how is airmap coming along? When I try to run /etc/airmap/AirMapSDK-Embedded/runtests.py, it fails by saying it cannot import module gps.

zehortigoza, 2017-10-26:
It would need to be implementation such emulation over MAVLink what I think would be really hard to do.

Polygonal geofencing don't need SDCard to work, I can do it with PX4 running in Aero. The problem may be because it is not available yet in QGC stable but you can do it with QGC daily(https://docs.qgroundcontrol.com/en/releases/daily_builds.html)

About rc.local/rc.txt you can always build the PX4 with your script, you will not be able to change in unless you compile it again.

lkm1321, 2017-11-14:
Thanks, got it working.

closed 2017-11-14 by zehortigoza