intel-aero/meta-intel-aero #287: FC flashing delay with 1.6

guermonprez, 2017-11-07:
hello

On my RTF unit, I noticed FC flashing with the new baud rate
coming with the 1.6 release, the looping flashing script may
have to run for a variable time, and potentially for a long time.
Up to 200 cycles (after 200 I stopped).
Typically 20-40 cycles. After several tries it always ended up working.

I updated the doc to warn users about the new potential delay. Other leads ?

thanks

lucasdemarchi, 2017-11-07:
> looping flashing script

what do you mean by that? the flashing LEDs?

lucasdemarchi, 2017-11-07:
btw, nothing changed in this release regarding this. So if the board got stuck, it applies to every release.  In the past it was because people had an older bios that was failing to reboot.  What bios version did you have before attempting the update?

guermonprez, 2017-11-07:
by looping script, I mean aerofc-update.sh

BIOS_VERSION = Aero-01.00.13
OS_VERSION = Poky Aero (Intel Aero linux distro) v1.6.0 (pyro)
AIRMAP_VERSION = 1.8
FPGA_VERSION = 0xc2
Aero FC Firmware Version = 1.6.5

lucasdemarchi, 2017-11-07:
Ok. I thought you were talking about the aero update process

zehortigoza, 2017-11-08:
Most of the time I update the firmware from PX4 and Ardupilot source code and I never had this problem but I can sometimes reproduce it by running the aerofc-update.sh script in Aero.

The only real difference between update from flight stack source code and from aerofc-update.sh script is the order of the baudrates. Just change the order don't look a good solution and could fail in future so a better fix would use the force-bootloader script.
Please replace you script with this one(https://github.com/intel-aero/meta-intel-aero-base/pull/23) and let me know if it also works for you

anitha-suresh, 2018-01-03:
Issue has been fixed with this [PR](https://github.com/intel-aero/meta-intel-aero-base/pull/22)

closed 2018-01-03 by anitha-suresh