intel-aero/meta-intel-aero #289: Bugfix: update Docker

JonasVautherin, 2017-11-10:
We are experiencing [this issue](https://github.com/moby/moby/issues/4861) when running our Aero without being connected to the Internet. This is a problem when we are testing in-field.

It seems like this has been fixed in Docker 17.10 (as mentioned in the link above).

Would it be possible to update Docker on the Aero Yocto?

JonasVautherin, 2017-11-23:
May I ask if this will be considered for a future update? :-)

guermonprez, 2017-11-23:
Hello

We do not plan to change docker on yocto for now.
You can use docker on Ubuntu too, but you’ll have to install it yourself.

Thanks

JonasVautherin, 2017-11-23:
You mean, running docker inside a docker container running Ubuntu? 

If yes, well, I don't think it is considered good practice to run docker inside docker. And it doesn't solve my problem since I will still be running the bugged docker on the Aero, right?

zehortigoza, 2017-11-24:
Now there is a Ubuntu PPA to install Aero packages into Ubuntu, look at the wiki for instructions.

JonasVautherin, 2017-11-24:
Ok, thank you!

closed 2017-11-24 by JonasVautherin