intel-aero/meta-intel-aero #290: [Feature] LTE connectivity through a free cloud VPN

AlexisTM, 2017-11-14:
Hey there;

I tested and can confirm that the LogMeIn Hamachi VPN (vpn.net) is working (x86, x64, ARMv7) to run ROS in the cloud. The free version allows 5 clients and the configuration is very easy. Just link the devices to your account then create a network with the UAVs you want to fly and your computer.

guermonprez, 2017-11-14:
Hello

Could you publish your notes and commands.
Are you using yocto or Ubuntu?

thanks

AlexisTM, 2017-11-14:
I am working on Ubuntu and it worked against lot of network architectures; LTE->LTE devices; Corporate network -> LTE, ...

I installed Hamachi via https://www.vpn.net/linux.

```bash
sudo hamachi login
sudo hamachi attach [my-email]
```

Then, from the internet (https://secure.logmein.com/central/Central.aspx) I create a Mesh network with the devices then I can use the static IPs provided by Hamachi.