intel-aero/meta-intel-aero #291: Cannot connect to flight controller and FPGA after Ubuntu installation

lukasvst, 2017-11-16:
We have installed Ubuntu on the Intel Aero RTF drone following the instruction at https://github.com/intel-aero/meta-intel-aero/wiki/90-(References)-OS-user-Installation

Although the flight controller and the FPGA have been working with Yocto before, after installing ubuntu calling sudo aero-get-version.py yields:
BIOS_VERSION = Aero-01.00.13
OS_VERSION = Ubuntu 16.04.03 LTS"
AIRMAP_VERSION = UNKNOWN
FPGA_VERSION = unknown
AeroFC firmware version = unknown

Also calling sudo systemctl restart mavlink-router gives an error message 
... (code=exited, status=1/FAILURE)

We have tried to update the flight controller using 
sudo aerofc-update.sh nuttx-aerofc-v1-default.px4
however this hangs with the message
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.

Can you help us with this issue?

zehortigoza, 2017-11-16:
Could you check if it booted with the right kernel version? It should be Linux 4.4.X

anselmolsm, 2017-11-16:
> AeroFC firmware version = unknown

It's a known issue, as mentioned at https://github.com/intel-aero/meta-intel-aero/wiki/90-(References)-OS-user-Installation#get-version . It's a script issue, not a communication problem.

lukasvst, 2017-11-21:
Thanks a lot for the answers! We were able to flash the flightcontroller now. The reason for the problem was that we had booted the wrong kernel version.

closed 2017-11-21 by lukasvst

srnetsr, 2018-02-15:
Hi guys, @zehortigoza @anselmolsm @lukasvst 
I got the same error, and check the kernel version.
The kernal version is
```
$ uname -a
Linux intel-aero-rtf-ubuntu 4.10.0-28-generic #32~16.04.2-Ubuntu SMP Thu Jul 20 10:19:48 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

```

srnetsr, 2018-02-15:
But I got the problem after that, @lukasvst @zehortigoza @anselmolsm .
I got the result from `sudo aero-get-version.py`
BIOS_VERSION = Aero-01.00.13
OS_VERSION = Ubuntu 16.04.03 LTS"
AIRMAP_VERSION = UNKNOWN
FPGA_VERSION = 0xc2
AeroFC firmware version = unknown

I flashed fc with
`$ sudo aerofc-update.sh nuttx-aerofc-v1-default.px4`
and flashed completely.
However, it is still `AeroFC firmware version = unknown `

Please help me...

anselmolsm, 2018-02-15:
aero-get-version.py calls a script called aerofc-get-version.py which needs pymavlink. Pymavlink is not packaged for Ubuntu (that's why it is not listed as a dependency). Install pymavlink (e.g. using pip).

srnetsr, 2018-02-15:
Hi @anselmolsm ,
thanks for your reply.

actually I did
`$ sudo pip install pymavlink`
by following [https://github.com/intel-aero/meta-intel-aero/wiki/90-(References)-OS-user-Installation#intel-aero-repository](url).

But there was some error. So I upgrade pip and tried
`$ sudo -H pip install pymavlink`

was that something wrong?
Do you have any idea?

anselmolsm, 2018-02-15:
The aforementioned script is written in python 3.
pip is for python 2. Try with pip3.

srnetsr, 2018-02-15:
I also installed python3-pip using
$ sudo apt-get install python3-pip

However, still I got the result from sudo aero-get-version.py as
BIOS_VERSION = Aero-01.00.13
OS_VERSION = Ubuntu 16.04.03 LTS"
AIRMAP_VERSION = UNKNOWN
FPGA_VERSION = 0xc2
AeroFC firmware version = **unknown**

anselmolsm, 2018-02-15:
See if aero**fc**-get-version.py outputs any helpful message and paste it here, please.

srnetsr, 2018-02-15:
I looked over aero-get-version.py, and I understand.
I am following https://github.com/intel-aero/meta-intel-aero/wiki/90-(References)-OS-user-Installation#get-version,
```
echo 'deb https://download.01.org/aero/deb xenial main' | sudo tee /etc/apt/sources.list.d/intel-aero.list
wget -qO - https://download.01.org/aero/deb/intel-aero-deb.key | sudo apt-key add -
sudo apt-get update
sudo apt-get upgrade
sudo apt-get -y install gstreamer-1.0 libgstreamer-plugins-base1.0-dev libgstrtspserver-1.0-dev gstreamer1.0-vaapi gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-libav ffmpeg v4l-utils python-pip
sudo pip install pymavlink
sudo apt-get -y install aero-system
sudo reboot
```
this is what I did.
But after run the above commanders, I cannot find any aerofc-get-version.py file in this computer of intel drone....

Do I need to install additional program?

anselmolsm, 2018-02-15:
It was to execute the command and get the output, no need to read the script.

Let's move this to https://github.com/intel-aero/packages/issues/37. Thanks.

srnetsr, 2018-02-16:
Now all issue in regard to this post is solved.

Follow the instruction in [https://github.com/intel-aero/meta-intel-aero/wiki/90-(References)-OS-user-Installation#get-version](url)

It works correctly now. Thanks.