intel-aero/meta-intel-aero #292: Fan control GPIO pin documentation

ChrisHughes, 2017-11-19:
Looking at the documentation on the Intel Aero board, it seems plausible that one can control the cpu cooling fan in software.

According to this document the fan is enabled by CPU_GPIO_4.
https://www.intel.com/content/dam/support/us/en/documents/boardsandkits/aero/intel-aero-compute-board-guide.pdf

And in this wiki, it lists hardware GPIO_4 as mapping to kernel GPIO 487.
```CPU_GPIO_04 GPIO-487```

However when I try to use the boilerplate code to control the fan state in the same way the on board LEDs are controlled, the entire board powers down abruptly.

 ```echo 487 > /sys/class/gpio/export
 echo out > /sys/class/gpio/gpio487/direction
 echo 0 > /sys/class/gpio/gpio487/value```

Can you please advise on the issue, and whether this is an issue in the documentation (i.e. wrong kernel gpio value), or the wrong way to control the fan state, or an expected behaviour / limitation of the hardware itself?

zehortigoza, 2017-11-20:
Hi Chris
It is actually the GPIO 485, you can use the bash script that is present in Yocto(and probably in Ubuntu) aero-fan-ctrl <on or off>

ChrisHughes, 2017-11-20:
Hi Zehortigoza,

Thanks for the heads up on the GPIO number. I can confirm this allows me to turn off the fan.

However when reenabling it, either by aero-fan-ctrl or echoing a value directly to the GPIO controller, the fan does not start spinning again, and the drone emits a high pitched whining noise. The fan is not trying to spin, and spinning the blades by hand confirms this. 

This behaviour was confirmed on Ubuntu 16.04, I have not tried on Yocto so far.

Any help on the issue would be appreciated. For noise reduction I am trying to write a daemon that monitors cpu temperature and enables the fan once a threshold is met.

zehortigoza, 2017-11-22:
Hey Chris
In all my drones I can turn off and turn on but it take a few seconds to the fan start to rotate again.

Are you trying to reduce the noise in the barometer? We flew with and without the fan(even off it can rotate depending on the wind) and there is no huge difference.

kuoshih, 2017-11-23:
Hi Zehortigoza and Chris,

Turn on: aero-fan-ctrl on
Turn off: aero-fan-ctrl off

You can install Psensors to monitor your CPU temperature.
Turning on the fan can cool down around 10~15 degree C of CPU on my RTF Drone. 
https://itsfoss.com/check-laptop-cpu-temperature-ubuntu/

ChrisHughes, 2017-11-23:
I've made a quick video showing the problem I'm having. Turning off the fan works fine, reading the cpu temps works fine. However turning back on the fan does not work, just produces a high pitched noise.

https://www.youtube.com/watch?v=xxHw5TxO5Xc

kuoshih, 2017-11-25:
"a high pitched noise" sounds like a PWM on the fan.
Could you use scope to measure the GPIO? 
or open the case of compute board and then check the wires of the fan?
I guess that it could be a pull high/low issue.

ChrisHughes, 2017-11-27:
It may be an issue with the fan itself. If the fan is unplugged from the power distribution board, then plugged back in, it will begin turning again. This indicates to me that the gpio switching works fine, but may have leakage current getting past the transistor that prevents the fan controller from shutting down.

I have also observed that the fan starts up again after about 15 seconds or so, so it appears it does a self-reset of some kind. I think given that behaviour I can live with the issue for now.

zehortigoza, 2017-11-27:
Yes there is some electrical problem preventing it to turn right away, so after 15 it works for you too!

closed 2017-11-27 by zehortigoza