intel-aero/meta-intel-aero #293: Ubuntu Installation on SSD

kuoshih, 2017-11-19:
Dear all,

After installing Yocto, I installed Ubuntu on 32GB MMC.
I selected "install Ubuntu alongside unknown Linux distribution."
(Yocto got 18GB and Ubuntu only got 9GB)
The information of 32GB MMC on Gparted is as follows: 
Partition             Name   Filesystem   Size
Dev/mmcblk0p1 Boot     FAT16          28  MB
Dev/mmcblk0p2              EXT3          18  GB
Dev/mmcblk0p3              EXT4          9.1 GB
Dev/mmcblk0p4 Swap    Linux-swap 1.45GB 

In this case, Ubuntu and Yocto worked well but Ubuntu needs more space.
I tried to install Ubuntu 16.04 on a 256GB SSD, which is plugged in M.2 connector 1 lane PCIe.
(SSD info. https://www.amazon.com/ADATA-SU800-M-2-256GB-ASU800NS38-256GT-C/dp/B01M3Z46FQ/ref=sr_1_3?s=electronics&ie=UTF8&qid=1511091975&sr=1-3&keywords=Adata+SSD+256GB+M.2+2280)
Then, I got some troubles in several cases.

[Case 1: Ubuntu on SSD]
After installing installing Yocto on MMC, I installed Ubuntu on 256GB SSD.
The information of 32GB MMC on Gparted is as follows: 
Partition     Name Filesystem Size
Dev/mmcblk0p1 Boot FAT16      28  MB
Dev/mmcblk0p2      EXT3       27  GB
Dev/mmcblk0p3 Swap Linux-swap 1.45GB 

After installing Ubuntu and rebooting, the system entered the grub command line.
I cannot enter Ubuntu or Yocto.
I tried to find the Ubuntu kernel and enter Ubuntu.
However, I cannot find HD information on SSD and only find HD information on MMC.

[Case 2: Ubuntu on MMC and LVM on SSD]
To extend Ubuntu space via LVM, the SSD is formated as Linux LVM format.
After installing Yocto and Ubuntu on 32GB MMC, the 9.1GB sapce is format as LVM format.
I selected "Something else" to install Ubuntu on 9.1GB space.
The information of 32GB MMC on Gparted is as follows: 
Partition     Name Filesystem Size
Dev/mmcblk0p1 Boot FAT16      28  MB
Dev/mmcblk0p2      EXT3       18  GB
Dev/mmcblk0p3      Linux LVM  9.1 GB
Dev/mmcblk0p4 Swap Linux-swap 1.45GB 

After installing Ubuntu and rebooting, the Ubuntu works well.
Then, the 9.1GB is extended to 261.1GB via VLM commands.
But, after rebooting Ububut, the system entered the grub command line again.
I found that if Ubuntu is relevant to SSD, the Yocto and Ubuntu will crash. 

Hence, my questions are as follows:
1. Is it a ePCI interface hardware bug?
2. Does anyone try to extend space using MicroSDcard?
3. Could I format MMC and install Ubuntu only, if I only want to use Ubuntu instead of Yocto?
If anyone has similar issues, please share your experience and solutions. 
Thanks!

guermonprez, 2017-11-20:
hello

yes you can install ubuntu as the only OS on the MMC. that's how I do it myself.
(I also had successful ubuntu-yocto dual boots in the past).

Note about the SD card: I think the SD card driver may not be loaded with the boot kernel, so I would not put my / system on this media (or have a spanned partition including the SD). you could use the SD card as additional file system, mounted at a later. I'm not sure, you can check that.

I have not tested a disk on the PCI. but I would keep it simple and boot from MMC, then have /home or similar folders on the SSD.

Let us know if/how the SSD works, we'll add that info to the wiki notes.

thanks

kuoshih, 2017-11-21:
Hi all,

I tried the following cases:

[Case 3: Ubuntu only on MMC]
After formating MMC, I installed Ubuntu on 32GB MMC.
The information of 32GB MMC on Gparted is as follows:
Partition Name Filesystem Size
Dev/mmcblk0p1 Boot FAT16 28 MB
Dev/mmcblk0p2 EX4 27 GB
Dev/mmcblk0p3 Swap Linux-swap 1.45GBEXT3

Ubuntu works well.

[Case 4: Ubuntu only on SSD]
After formating MMC and SSD, I installed Ubuntu on 256GB MMC.
After Ubuntu installation and reboot, the system cannot enter Ubutntu and only enter "UEFI interactive shell."

[Case 5: Ubuntu on MMC and LVM on SSD]
After formating MMC and SSD, I installed Ubuntu on 32GB MMC.
Ubuntu works well.
Them, 26GB MMC is extended to 260.9GB via VLM commands.
Everything works perfectly.
However, after installing Intel Aero repository and rebooting, the Ubuntu cannot start and stuck
"loading initial ramdisk ..." 

Anyway, I use Case 3 "Ubuntu on MMC" and mount SSD as a media.
I am not sure it's a PCIe interface bug or a fault of SSD.

kuoshih, 2017-11-21:
In case someone needs to install SSD on RTF.
The installation of SSD is as follows:
1.1 Unscrew the case of RTF.
1.2 Open the case 
1.3 Unplug connector  9th & 15th connectors (You can unplug 5th,6th, and 7th connector to turn the board easily)
[image]
2 Plugin SSD at  13th connector M.2 inteface PCIe for SSD
[image]
3. Before putting the board, you should check 5th connector is still plugged or not since it was blocked by a black tape. You should also plug 15th connector GND
[image]
Then, you can enter Ubuntu and use Gparted to format SSD.