intel-aero/meta-intel-aero #294: Front and bottom (optical flow) cameras data stream in ROS

jsfurtad, 2017-11-24:
Hello,

I am running Ubuntu 14.04 on the Intel-aero RTF drone with ROS installed after following the steps on the wiki. I have no problem with the realsense but how do I obtain the front and bottom facing camera data as rostopics?

Thanks,
Joshua Furtado

zehortigoza, 2017-11-27:
There is no official ROS node to access the Aero cameras, those cameras can be accessed by video4linux APIS so maybe there is a generic node that you can use.

Front camera: /dev/video2 id=0 example: https://github.com/intel-aero/sample-apps/tree/master/hd-camera
We just enabled this camera, the image quality is not good you will need to configure focus, white balance, exposure to get a good image.

Bottom camera: /dev/video id=1 example: https://github.com/intel-aero/aero-optical-flow

closed 2018-06-05 by zehortigoza

danielemarchetti, 2018-07-30:
Hi,
I tried the example https://github.com/intel-aero/sample-apps/tree/master/hd-camera to acquire images from the frontal camera. When I run:
$ g++ -o hd-camera.bin hd-camera.cpp -lopencv_core -lopencv_imgproc -lopencv_highgui -std=c++11
The application is executed without error but return after a while without showing the images from the camera. Can you help me in some way ?
Thank you in advance.
Daniele

AlexisTM, 2018-09-05:
@jsfurtad 
Bumping into the same problem, I made this node: https://github.com/AlexisTM/aero-optical-flow-ros